Ysurac/openmptcprouter #1992: Are there any ways to run openmptcprouter and raspberry os at the same time? (Raspberry Pi)

zxwanghongyi, 2021-07-29:
I want to run some else, not support openwrt, with aggregation in only one raspberry pi.
Are there any ways to run openmptcprouter and raspberry os at the same time?
Or is it possible to install all the things that need to run aggreation in linux without using openwrt?
Thanks!

Ysurac, 2021-07-30:
You can't run OpenMPTCProuter ate the same time as raspberry OS.
I've made a long time ago a tutorial for RPI: http://mptcp.zugaina.com/ but it's really outdated now.

zxwanghongyi, 2021-07-30:
Thanks! what about raspberry os with docker running openmptcprouter?

zxwanghongyi, 2021-07-30:
Are there any tutorials of adding mptcp support to latest kernel you suggest?

Ysurac, 2021-07-30:
Host OS must have a patched kernel with MPTCP support even in Docker. I never tested this.
You can find the MPTCP patch I use here: https://github.com/Ysurac/openmptcprouter/blob/develop/root/target/linux/generic/hack-5.4/690-mptcp_trunk.patch
This may apply to a 5.4 RPI kernel.
I don't know any tutorials for that.

zxwanghongyi, 2021-07-30:
Thank you! I'm gona try that

zxwanghongyi, 2021-07-31:
Another question, I'm following http://mptcp.zugaina.com/ and your patch, It's really helpful. I success to get MPTCP's kernel running on raspberry os, but installing MCTCP on vps is bothering me now.
In your old tutorial, getting MCTCP run on vps only need an apt-get install, but the repository server is not avalible any more.
There are few deb packges in https://github.com/multipath-tcp/mptcp/releases , do I need to install them all in orders?
Cause after I install them all, shadowsocks still saying  failed to enable multipath TCP: Protocol not available. Thanks for the help!

Ysurac, 2021-07-31:
You only need linux-image, check that it's in grub boot menu and reboot. You should have it installed, check with ```uname -a```

zxwanghongyi, 2021-08-01:
Thanks! I success to get both vps and rpi's mctcp work, but the vps's shadowsocks keep showing 
`2021-08-02 01:38:38 INFO: [6466] connect to 127.0.0.1:1080
 2021-08-02 01:38:38 INFO: new connection to remote, 1 opened remote connections
 2021-08-02 01:38:38 ERROR: getpeername: Transport endpoint is not connected
 2021-08-02 01:38:38 INFO: close a connection to remote, 0 opened remote connections
 2021-08-02 01:38:38 INFO: close a connection from client, 0 opened client connections
`
when any internet access comming from raspberry pi
I already tried disabled firewall, change both remote and local port
I'm using apt install shadowsocks-libev to install shadowsocks.

Ysurac, 2021-08-01:
No idea what release is available on Debian. You should try to have same release on both end.

zxwanghongyi, 2021-08-02:
Now I compile and install same version of shadowsocks in two machine, still showing that, dose "Config Network" part effect this?
1 Do I need to set proxy to 127.0.0.1:1080 in order to use aggregation?
2 Do I need to follow http://mptcp.zugaina.com/ [Network] part if I only need to use aggregation in raspberry pi (not as a router)?
This is my current setup: 4g moudle config as a network card(usb0)-internet, usb to rj45 (eth1)-internet, raspberry pi rj45(eth0)-my computer.
`
auto lo
iface lo inet loopback
auto eth0
iface eth0 inet static
 address 192.168.100.1
 netmask 255.255.255.0
 
auto usb0:line1
iface usb0:line1 inet static
 address 192.168.225.29
 netmask 255.255.255.0
 gateway 192.168.225.1
 

auto eth1:line2
iface eth1:line2 inet static
 address 192.168.2.71
 netmask 255.255.255.0
 gateway 192.168.2.1
 post-up /opt/mptcp/rules.sh
`
`
ip rule add from 192.168.225.29 table 1
ip rule add from 192.168.2.71 table 2
ip route add 192.168.225.0/24 dev usb0:line1 scope link table 1
ip route add default via 192.168.225.254 dev usb0:line1 table 1
ip route add 192.168.2.0/24 dev eth1:line2 scope link table 2
ip route add default via 192.168.2.254 dev eth1:line2 table 2
ip route add default scope global nexthop via 192.168.2.254 dev usb0:line1

ss-nat -s 152.70.109.118 -l 1080 -u
`
3 Are there any other ways to use aggregation rather than shadowsocks? 
I notice that in [wiki/technologies/How this work ?](https://github.com/Ysurac/openmptcprouter/wiki/Technologies) There are some ways dont even need mptcp in kernal, may be easier to use that way rather than mptcp+shadowsocks?

closed 2021-11-05 by github-actions[bot]