Ysurac/openmptcprouter #1993: VPS multiple IP setup

SonicSteel, 2021-07-30:
## Expected Behavior
Attempting to set up multiple IP’s as shown in https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs

## Current Behavior
I am unsure of what method to use to get this option to come up automatically as shown in the wiki. I have a VPS hosted on OVH which I can assign multiple IP’s to.  Or if this can be set up manually.
VPS has a single NIC, /etc/network/interfaces is set for eth0 and eth0:0 to have different public IP’s
It is currently working great (though a bit slow) with a single IP.

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: current Debian 10 x86_64 version (pulled within last 12 hours)
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-07-30:
IPs need to be configured before omr-admin is started on the VPS, so at boot. Then you can save&apply wizard again to get all you need on the router.

SonicSteel, 2021-07-30:
I did a fresh install all around, this time configuring the additional IP's before running the VPS script.
Currently there are 10 additional IP's on the VPS (0.1026 5.4.100-mptcp)
I did the reboot after the script, connected a fresh install of OMR (0.58.5) to the VPS, and shadowsocks still only shows the default single rule
Remote Servers lists the same server twice as sss0 (enabled) and sss1 (disabled)

Ysurac, 2021-07-30:
I will make tests.

SonicSteel, 2021-07-30:
Just to add, I'm not sure if it's making the difference, but I have ifupdown2 installed.

I also tried 3 methods (separately) of attaching the IP's via /etc/network/interfaces:

```
iface eth0 inet static
    address x.x.x.x/32
    gateway x.x.x.x
    up   ip addr add x.x.x.x/32 dev $IFACE label $IFACE:0
    down ip addr del x.x.x.x/32 dev $IFACE label $IFACE:0
    up   ip addr add x.x.x.x/32 dev $IFACE label $IFACE:1
    down ip addr del x.x.x.x/32 dev $IFACE label $IFACE:1
```

```
auto eth0
allow-hotplug eth0
iface eth0 inet static
    address x.x.x.x/32
    gateway x.x.x.x

auto eth0:0
allow-hotplug eth0:0
iface eth0:0 inet static
    address x.x.x.x/32
```
    
  
```
auto eth0
allow-hotplug eth0
iface eth0 inet static
    address x.x.x.x/32
    gateway x.x.x.x

iface eth0 inet static
    address x.x.x.x/32

iface eth0 inet static
    address x.x.x.x/32
```

Ysurac, 2021-07-30:
All IPs are public IPs ?

SonicSteel, 2021-07-30:
Yes, all IP’s are public, and pingable when configured.
They are the additional IP’s you can order through OVH

SonicSteel, 2021-08-04:
Just an update, I started using the image for 0.56.5 and it was working as expected.  I'll also do some additional testing with different versions.

Ysurac, 2021-08-07:
I just tested on latest release on a fresh install and it's working as expected.

SonicSteel, 2021-08-07:
I’ll give it another shot, having some trouble now with shadowsocks not starting
Is there a limit to additional IP’s that I’m passing maybe?

SonicSteel, 2021-08-08:
With an excessive number of IP's, it seems to cause problems with the shadowsocks config.
A fresh install of 0.58.5 still doesn't pick up the additional IP's (testing with 17 total), but connecting 0.57.3 to the same VPS and it appears to work for me.
I'm not sure what I'm doing differently than you are, but unless there's features or stability in 0.58.5 that I'm missing out on, I think this will work for me for now.

closed 2021-11-11 by github-actions[bot]