Ysurac/openmptcprouter #1994: VPS Server on Google Cloud

netajeepatil, 2021-07-30:
Hi,
I am hosted vps server on google cloud and opened port 0-65535 tcp&? udp in firewall. my mptcp router is shows connectivity , when i telnet  suggested ports on vps config, all are reachable. When i am checking amiusingmultipath website it shows no multi-path not enabled. please help to sort this.

Ysurac, 2021-07-30:
```uname -a``` on VPS show mptcp in kernel name ?
Aggregation is working or not ?
What is the result of ```curl http://www.multipath-tcp.org``` from VPS ?

netajeepatil, 2021-07-31:
nikunjkaria@linuxvps:~$ uname -a
Linux linuxvps 5.8.0-1038-gcp #40~20.04.1-Ubuntu SMP Thu Jul 15 11:55:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
nikunjkaria@linuxvps:~$ curl http://www.multipath-tcp.org
Nay, Nay, Nay, your have an old computer that does not speak MPTCP. Shame on you!

Aggregation is working or not ? - how to check this , on mptcp it able to connect with vps ip , when checks for speed download speed is good but issue with upload speed is not upto the mark.

netajeepatil, 2021-07-31:
my system details on GCP as below - Machine type - n1-standard-1 (1 vCPU, 3.75 GB memory), 20GB Balanced speedHDD
CPU platform - Intel Skylake
Zone - asia-south1-a
image - ubuntu-2004-focal-v20210720
HDD - 20 GB Balanced persistent disk

Ysurac, 2021-07-31:
The kernel is not the MPTCP patched kernel. Did you reboot the VPS ?

netajeepatil, 2021-07-31:
yes after install rebooted

Ysurac, 2021-07-31:
I only tested Debian on GCP. Try to install Debian and launch script again.

netajeepatil, 2021-07-31:
ok, will try this also.  I tried this on GCP 4 months ago and worked fine & when I am trying for second time its not working. so I raise query. how will We get know that kernel is n MPTCP patched kernel.

Ysurac, 2021-07-31:
When you do a ```uname -a``` on VPS you should get a text that contain something like "Linux vps 5.4.100-**mptcp** ...".

netajeepatil, 2021-08-02:
Debian worked with GCP. Tested Ubuntu and worked with AWS.
Thanks for support.

closed 2021-08-02 by netajeepatil