Ysurac/openmptcprouter #1995: Swap is not working

finnschi, 2021-07-30:
## Expected Behavior

swapon is working

## Current Behavior
swapon is not working

no logs
## Possible Solution
enable kernel support for swap?

## Steps to Reproduce the Problem
1.use fdisk to create a swap aprtition /dev/sdX
2. mkswap /dev/sdX 
3. swapon /dev/sdX

`block: failed to swapon /dev/sdaX (-1)`

## Context (Environment)

WEb interface crashes a lot, googeling said to activate swap 

I want to try to enable swap, and thats not working for me

## Specifications

  - OpenMPTCProuter version: v0.58.5 
  - OpenMPTCProuter VPS version: 0.1026.5.4.100-mptcp 
  - OpenMPTCProuter platform: X86_64

Ysurac, 2021-07-30:
Kernel support swap.
You may need swap only if memory is fully used.

finnschi, 2021-08-05:
ok .. hmm ive got way enough ram (8G) still interface crashes and doesnt come until I reboot the appliance

Ysurac, 2021-08-05:
Check system log to find why the interface can crash.

finnschi, 2021-08-05:
cool, I didnt see anything in the logs but probably just looked at the wrong place, I can recreate the UI crash so if you could tell me what Log I should be looking at I can see why it crashes :)

Ysurac, 2021-08-05:
Should be in ```logread``` via SSH (or State->System log via interface but when crashed it's more difficult)

closed 2021-11-09 by github-actions[bot]