Ysurac/openmptcprouter #1997: Raspberry Pi4 AP problem

highfreq, 2021-08-01:
If you connect once as a client to an access point than it becomes impossible to switch off client mode and restart master. Master interface after few seconds, even if it is configured as AP, goes back to client mode. 
I think it is a bug.

closed 2021-11-04 by github-actions[bot]

Adorfer, 2021-11-05:
note: This is an upstream issue on openwrt for PI4. 

And if you ask at OpenWRT you will probably get the answer that the broadcom driver is not real open source (as far as i understood) and that it's not totally stable if you want to do more than just ap-client ("station mode").

highfreq, 2021-11-05:
Actually never had a problem after running rpi-eeprom-update. It works flawlessy now.

Adorfer, 2021-11-05:
O.k. then it was an upstream issue not on OpenWRT but at RPI project. 
(BTW: It would have been great if you had posted the found solution here yourself AND closed your own issue!)

highfreq, 2021-11-05:
Actually it just started working and i forgot about it. I wasn't looking for a solution for that problem when i updated.