Ysurac/openmptcprouter #1998: Help with Firewall Rules not working.

Yersi88, 2021-08-01:
I want to create firewall rules to block internet access to some devices (Smart Lightbulbs) I only want them to be able to access LAN, and control them on LAN, but not report back home to China. I tried to create this rules: 

[image]
[image]

But unfortunately they don't work. I'm still able to control the lights from the app through the internet. I tried to include my PC's IP in those rules but it too still had access to internet. 

I'm not sure what I'm doing wrong, seems to me like it should just work. 

## Specifications

  - OpenMPTCProuter version: 0.58.4 
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: x86_64

Any help much appreciated :) Sorry if the question is dumb/unrelated to mptcp

Ysurac, 2021-08-01:
Drop should be set as traffic rules on forward zone.

Yersi88, 2021-08-01:
> Drop should be set as traffic rules on forward zone.

[image]
Is this what you mean? 

Doesn't work as well :(

Ysurac, 2021-08-01:
Yes, but I'm not sure that I made this working with multiples IPs...

Yersi88, 2021-08-01:
It's not working even if I put one IP there instead of 8 :(

Ysurac, 2021-08-01:
I would need the result of ```iptables-save```

Yersi88, 2021-08-02:
> I would need the result of `iptables-save`

`root@OpenMPTCProuter:~# iptables-save
# Generated by iptables-save v1.8.7 on Mon Aug  2 11:22:02 2021
*raw
:PREROUTING ACCEPT [943420928:644775129062]
:OUTPUT ACCEPT [769840682:594771031576]
COMMIT
# Completed on Mon Aug  2 11:22:02 2021
# Generated by iptables-save v1.8.7 on Mon Aug  2 11:22:02 2021
*nat
:PREROUTING ACCEPT [6626:1794341]
:INPUT ACCEPT [2542:183813]
:OUTPUT ACCEPT [13798:1029697]
:POSTROUTING ACCEPT [2394:168796]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:v2r_def_dst - [0:0]
:v2r_def_forward - [0:0]
:v2r_def_local_out - [0:0]
:v2r_def_pre_src - [0:0]
:v2r_def_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -p tcp -j v2r_def_pre_src
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i wan1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i usb0 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i usb1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j v2r_def_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o wan1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o usb0 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o usb1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A v2r_def_dst -m mark --mark 0x539 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A v2r_def_dst -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x5391200/0xffffffff
-A v2r_def_dst -m mark --mark 0x5391200 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_usb0 dst -j MARK --set-xmark 0x5394/0xffffffff
-A v2r_def_dst -m mark --mark 0x5394 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_usb1 dst -j MARK --set-xmark 0x5397/0xffffffff
-A v2r_def_dst -m mark --mark 0x5397 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_wan1 dst -j MARK --set-xmark 0x5393/0xffffffff
-A v2r_def_dst -m mark --mark 0x5393 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_eth0 dst -j MARK --set-xmark 0x5396/0xffffffff
-A v2r_def_dst -m mark --mark 0x5396 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_lo dst -j MARK --set-xmark 0x5395/0xffffffff
-A v2r_def_dst -m mark --mark 0x5395 -j RETURN
-A v2r_def_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_bypass dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_forward dst -j v2r_def_forward
-A v2r_def_dst -m comment --comment "dst_default: forward" -j v2r_def_forward
-A v2r_def_forward -p tcp -j REDIRECT --to-ports 1897
-A v2r_def_local_out -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
[... 312 lines cut ...]
-A zone_wan_dest_ACCEPT -o usb0 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o usb0 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o usb1 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o usb1 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o usb1 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i usb1 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Mon Aug  2 11:22:02 2021`

Ysurac, 2021-08-02:
My block script only work when tcp proto is specified, I should fix that...

Ysurac, 2021-08-02:
Should work when tcp & udp or only tcp is selected.

Yersi88, 2021-08-02:
> Should work when tcp & udp or only tcp is selected.

Odd, this is _not_ working for me: (PC still able to reach internet) 
[image]

Ysurac, 2021-08-02:
Try to only set TCP, and again do a "iptables-save".
The proxy used is shorewall ? I never tested block with v2ray.

Yersi88, 2021-08-02:
> Try to only set TCP, and again do a "iptables-save".
> The proxy used is shorewall ? I never tested block with v2ray.

I use v2ray as a proxy. 

```root@OpenMPTCProuter:~# iptables-save
# Generated by iptables-save v1.8.7 on Mon Aug  2 14:42:02 2021
*raw
:PREROUTING ACCEPT [951396876:649720012972]
:OUTPUT ACCEPT [777353692:599435288257]
COMMIT
# Completed on Mon Aug  2 14:42:02 2021
# Generated by iptables-save v1.8.7 on Mon Aug  2 14:42:02 2021
*nat
:PREROUTING ACCEPT [3358:898626]
:INPUT ACCEPT [1625:110347]
:OUTPUT ACCEPT [7405:554708]
:POSTROUTING ACCEPT [1386:99705]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:v2r_def_dst - [0:0]
:v2r_def_forward - [0:0]
:v2r_def_local_out - [0:0]
:v2r_def_pre_src - [0:0]
:v2r_def_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -p tcp -j v2r_def_pre_src
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i wan1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i usb0 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i usb1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j v2r_def_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o wan1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o usb0 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o usb1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A v2r_def_dst -m mark --mark 0x539 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A v2r_def_dst -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x5391200/0xffffffff
-A v2r_def_dst -m mark --mark 0x5391200 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_usb0 dst -j MARK --set-xmark 0x5394/0xffffffff
-A v2r_def_dst -m mark --mark 0x5394 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_usb1 dst -j MARK --set-xmark 0x5397/0xffffffff
-A v2r_def_dst -m mark --mark 0x5397 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_wan1 dst -j MARK --set-xmark 0x5393/0xffffffff
-A v2r_def_dst -m mark --mark 0x5393 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_eth0 dst -j MARK --set-xmark 0x5396/0xffffffff
-A v2r_def_dst -m mark --mark 0x5396 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_lo dst -j MARK --set-xmark 0x5395/0xffffffff
-A v2r_def_dst -m mark --mark 0x5395 -j RETURN
-A v2r_def_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_bypass dst -j RETURN
-A v2r_def_dst -m set --match-set ssr_def_dst_forward dst -j v2r_def_forward
[... 315 lines cut ...]
-A zone_wan_dest_ACCEPT -o usb0 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o usb0 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o usb1 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o usb1 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o usb1 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i usb1 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Mon Aug  2 14:42:02 2021```