Ysurac/openmptcprouter #2000: Emulate OpenMPTCProuter on RPI4 using qemu-system-arm

jjak0b, 2021-08-01:
I have installed a raspberry PI OS 64bit running on a RPI4 I'm trying to figure out how to configure qemu-system-arm using the precompiled images.
The provided precompiled images should have kernel and rootfs combined, but how i should set the kernel and the drive images in qemu from them ?

Ysurac, 2021-08-01:
Images are full disk images. Rootfs and kernel images are also availables in downloads directory.
But using OpenMPTCProuter on RPI4 VM is really not a good idea: a RPI4 doesn't have a so good CPU...

jjak0b, 2021-08-01:
About CPU i know i won't get the best performances but i would like to try it.
I'm a noob on virtualization stuff but i'm still learning.
However i can't see separated rootfs and kernel images on downloads directory: I can only see the sysupgrade and factory images in their ext4 and squashfs versions [here](https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/). 
I should use the sysupgrade image when i already have an OpenMPTCProuter installation,
and i should use the factory image to mount it on the sd card ( on non virtualized scenario) from what i understanded. Am i missing something ?

So i should set the ext4-factory image as drive ( as qemu parameter `-drive file=openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2709-rpi-2-ext4-factory.img,format=raw`  ) but what i should use as kernel parameter ( `-kernel $IMAGE -append "root=/dev/sda2" ` ) ?

this is the script i would use to run the qemu vm:
```
#!/bin/sh

MACHINE=raspi2
RAM=512
KERNEL=??
DRIVE=openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2709-rpi-2-ext4-factory.img
EMU32=qemu-system-arm

$EMU32 -M $MACHINE -m $RAM \
-kernel $KERNEL -append "root=/dev/sda2" \
-drive file=$DRIVE,format=raw
```
i followed the [openwrt guide on arm](https://openwrt.org/docs/guide-user/virtualization/qemu#openwrt_in_qemu_arm) but as the guide shows i expected to find a zImage in the downloads directory to use as kernel image

Ysurac, 2021-08-01:
True rootfs is only generated for x86_64 and can be used for Docker.
In you case you should use full image as it's a complete virtualisation. kernel is already in the raw image, you don't need to specify an external kernel.
So should be something like ```qemu-system-arm -m 1024 -machine type=raspi3 -drive format=raw,file=openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2709-rpi-2-ext4-factory.img``` or something like that.

jjak0b, 2021-08-03:
Setting the drive as the combined image doesn't boot on qemu.
[here for openwrt](https://gist.github.com/extremecoders-re/f2c4433d66c1d0864a157242b6d83f67#file-openwrt-qemu-md) says that the combined image as initramfs should works if set as kernel. I tried using the combined image of OpenMPTCProuter but again doesn't boot. 
So i though to use the openwrt armvirt kernel and it worked:
```

EMU32=/home/pi/qemu-6.0.0/build/qemu-system-arm #emu-system-arm
EMU64=/home/pi/qemu-6.0.0/build/qemu-system-aarch64 #qemu-system-aarch64
EMU=$EMU32
MACHINE=virt
RAM=512m
KERNEL=./omr32/openwrt-19.07.7-armvirt-32-zImage
DRIVE=./omr32/openmptcprouter-rpi-2-ext4-factory.img

$EMU  \
  -M $MACHINE \
  -m $RAM \
  -kernel $KERNEL \
  -append "root=/dev/vda2 panic=1" \
  -drive "file=${DRIVE},if=none,index=0,media=disk,format=raw,id=disk0" \
  -device "virtio-blk-pci,drive=disk0,disable-modern=on,disable-legacy=off" \
  -device virtio-net-pci \
  -netdev user,id=net1,hostfwd=tcp::2222-:22 \
  -device virtio-net-pci,netdev=net1 \
  -no-reboot -nographic
```
but i don't think si the right way to do that.

( i resized always the combined image because qemu errors:
```
FILE=openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2709-rpi-2-ext4-factory
TOFILE=openmptcprouter-rpi-2-ext4-factory

$HOME/qemu-6.0.0/build/qemu-img convert -f raw -O raw "$FILE.img" "$TOFILE.img"
$HOME/qemu-6.0.0/build/qemu-img resize -f raw "$TOFILE.img" 1G
```
)
So i tried this [qemu-rpi-kernel](https://github.com/dhruvvyas90/qemu-rpi-kernel/tree/master/native-emulation) and worked for raspianOS.
I tried the same extracting kernel.img and dtb ( as says also [here](https://github.com/dhruvvyas90/qemu-rpi-kernel/issues/127#issuecomment-821532701) ) and setting the drive as combined image but doesn't boot.

i tried all this with both arm and aarch64 qemu-systems in version 3.10 and 6.0.0

jjak0b, 2021-08-03:
However using the openwrt kernel for armvirt i think a lot of things are missing, for example i have no ethernet vitual interface if i use ifconfig. This is the output:
```
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[   13.484797] mount_root: mounting /dev/root
[   13.507552] EXT4-fs (vda2): warning: mounting unchecked fs, running e2fsck is recommended
[   13.555727] EXT4-fs (vda2): re-mounted. Opts: (null)
[   13.560652] mount_root: loading kmods from internal overlay
[   13.718168] kmodloader: no module folders for kernel version 4.14.221 found
[   13.731904] mount_root: failed to launch kmodloader from internal overlay
[   14.166205] block: attempting to load /etc/config/fstab
[   14.239975] block: extroot: not configured
[   14.272318] urandom-seed: Seed file not found (/etc/urandom.seed)
[   15.650177] procd: - early -
[   17.227658] procd: - ubus -
[   17.612162] urandom_read: 1 callbacks suppressed
[   17.615671] random: ubusd: uninitialized urandom read (4 bytes read)
[   17.794628] random: ubusd: uninitialized urandom read (4 bytes read)
[   17.848621] random: ubusd: uninitialized urandom read (4 bytes read)
[   17.995865] procd: - init -
Please press Enter to activate this console.
[   25.326332] random: crng init done
[   25.334869] random: 3 urandom warning(s) missed due to ratelimiting
[   32.103841] kmodloader: no module folders for kernel version 4.14.221 found

(none) login: root
login[464]: root login on 'ttyAMA0'

BusyBox v1.33.0 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+16336-b36068d35d)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58.5
 TARGET:      rpi2
 ARCH:        arm_cortex-a7_neon-vfpv4

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sat Jul 10 17:03:29 UTC 2021
-----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenMPTCProuter:~# ifconfig
root@OpenMPTCProuter:~# 
```

Ysurac, 2021-08-04:
You can't use OpenWrt kernel as at least MPTCP patch is missing.

jjak0b, 2021-08-05:
I supposed that, but it's weird that even using the openmptcprouter's kernel inside the  combined image doesn't boot.
I got and edited the script from [here](https://github.com/dhruvvyas90/qemu-rpi-kernel/tree/master/native-emulation) using the openmptcprouter's kernel, rootfs and dtb
```
EMU32=/home/pi/qemu-6.0.0/build/qemu-system-arm #emu-system-arm
EMU64=/home/pi/qemu-6.0.0/build/qemu-system-aarch64 #qemu-system-aarch64
EMU=$EMU32
MACHINE=raspi2b
RAM=1G
DRIVE="./omr32/openmptcprouter-rpi-2-ext4-factory.img"
KERNEL="./omr32/kernel.img"
DTB="./omr32/bcm2710-rpi-2-b.dtb"

$EMU  \
  -M $MACHINE \
  -m $RAM \
  -smp 4 \
  -append "rw earlyprintk loglevel=8 console=ttyAMA0,115200 dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootdelay=1" \
  -dtb "${DTB}" \
  -drive file=$DRIVE,if=sd,format=raw  \
  -kernel "${KERNEL}" \
  -nographic
  # -serial stdio -usb -device usb-mouse -device usb-kbd \
```

closed 2021-11-09 by github-actions[bot]

jjak0b, 2022-03-10:
i got some "results" ... upgrading qemu to 6.2.0 and running the following script with qemu, something changed.
(before running this i run only the qemu-img resize code)

```
$HOME/qemu-6.2.0/build/qemu-system-aarch64 \
	-M raspi3b \
	-cpu cortex-a53 \
	-m 1G \
	-serial stdio \
	-drive file=openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2710-rpi-3-ext4-factory.img,format=raw,if=sd \
	-no-reboot \
	-dtb bcm2710-rpi-3-b-plus.dtb \
	-kernel kernel8.img \
	-append "earlyprintk loglevel=8 console=ttyAMA0 dwc_otg.lpm_enable=0 earlycon=pl011,0x3f201000 root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait" \
	-netdev type=user,id=net,hostfwd=tcp::2222-:22 \
	-device usb-net,netdev=net \

```

Qemu seems to boot but stop the emulation after running /sbin/init. Here the log:

```
VNC server running on 127.0.0.1:5900
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.106 (runner@fv-az206-57) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16336-b36068d35d)) #0 SMP Thu Mar 25 05:03:24 2021
[    0.000000] Machine model: Raspberry Pi 3 Model B+
[    0.000000] earlycon: pl11 at MMIO 0x000000003f201000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000038000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 245760
[    0.000000]   DMA zone: 3840 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 245760 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45208 r8192 d32616 u86016
[    0.000000] pcpu-alloc: s45208 r8192 d32616 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 241920
[    0.000000] Kernel command line: earlyprintk loglevel=8 console=ttyAMA0 dwc_otg.lpm_enable=0 earlycon=pl011,0x3f201000 root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 886724K/983040K available (7038K kernel code, 414K rwdata, 1820K rodata, 512K init, 536K bss, 30780K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 62.50MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1cd42e208c, max_idle_ns: 881590405314 ns
[    0.000136] sched_clock: 56 bits at 62MHz, resolution 16ns, wraps every 4398046511096ns
[    0.009811] Console: colour dummy device 80x25
[    0.011756] Calibrating delay loop (skipped), value calculated using timer frequency.. 125.00 BogoMIPS (lpj=250000)
[    0.012365] pid_max: default: 32768 minimum: 301
[    0.014750] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.015092] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.040089] Disabling memory control group subsystem
[    0.068499] ASID allocator initialised with 32768 entries
[    0.069765] rcu: Hierarchical SRCU implementation.
[    0.077384] smp: Bringing up secondary CPUs ...
[    0.081202] Detected VIPT I-cache on CPU1
[    0.082034] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.087115] Detected VIPT I-cache on CPU2
[    0.087228] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.089380] Detected VIPT I-cache on CPU3
[    0.089502] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.090050] smp: Brought up 1 node, 4 CPUs
[... 491 lines cut ...]
[    1.274713] sdhost-bcm2835 3f202000.mmc: GPIO lookup for consumer wp
[    1.274978] sdhost-bcm2835 3f202000.mmc: using device tree for GPIO lookup
[    1.275285] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node '/soc/mmc@7e202000[0]'
[    1.275661] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node '/soc/mmc@7e202000[0]'
[    1.276035] sdhost-bcm2835 3f202000.mmc: using lookup tables for GPIO lookup
[    1.276328] sdhost-bcm2835 3f202000.mmc: No GPIO consumer wp found
[    1.324896] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    1.325856] of_cfs_init
[    1.328064] of_cfs_init: OK
[    1.337250] uart-pl011 3f201000.serial: no DMA platform data
[    1.340750] Waiting for root device /dev/mmcblk0p2...
[    1.378163] mmc0: host does not support reading read-only switch, assuming write-enable
[    1.379559] mmc0: new high speed SD card at address 4567
[    1.380335] mmcblk mmc0:4567: no of_node; not parsing pinctrl DT
[    1.383838] mmcblk0: mmc0:4567 QEMU! 1.00 GiB
[    1.397639]  mmcblk0: p1 p2
[    1.441759] EXT4-fs (mmcblk0p2): warning: mounting unchecked fs, running e2fsck is recommended
[    1.448618] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
[    1.449340] VFS: Mounted root (ext4 filesystem) on device 179:2.
[    1.455594] devtmpfs: mounted
[    1.460570] Freeing unused kernel memory: 512K
[    1.465245] Run /sbin/init as init process
[    1.994843] init: Console is alive

```

jjak0b, 2022-03-10:
i forgot to say that i tested this not using native emulation, because i want to test it as vm before using it on rpi4.
the vm host is not a rpi4 but a x86_64 machine, but should't be a problem since i'm emulating rpi

jjak0b, 2022-03-10:
After some tests i find out that it's an OpenWrt issue.
for some reason i think the built OpenWrt image isn't qemu compatible ( maybe because there are network issues about virtio-net-pci).
The only way that i tested  and works is setting the machine type as `virt` and set the kernel & fsroot in a drive both of a compiled OpenMPTCRouter build for armvirt platform.
The build process is explained [here](https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform); Just checkout to a version and set the same version to OMR_FEED_SRC. in my case i set `OMR_TARGET="armvirt" OMR_FEED_SRC=" v0.58.5"`.
In `make menuconfig` phase, i set the  arm virt target and cortex-a53 architecture.
After the build the images are in the `armvirt/source/bin/` folder.
In my case i run the following code:
```
EMU32=qemu-system-arm
EMU64=qemu-system-aarch64
EMU=$EMU64
MACHINE=virt
CPU=cortex-a53 # 64bit
RAM=1G
DRIVE="openmptcprouter-v0.58.5-f557cb00-r0+16336-b36068d35d-armvirt-64-rootfs-ext4.img"
KERNEL="openmptcprouter-v0.58.5-f557cb00-r0+16336-b36068d35d-armvirt-64-Image"

$EMU \
	-M virt \
	-cpu $CPU \
	-m $RAM \
	-no-reboot -nographic \
	-drive file=$DRIVE,format=raw,if=virtio \
	-kernel $KERNEL \
	-append "earlyprintk loglevel=8 console=ttyAMA0 root=/dev/vda rootfstype=ext4 rootwait" \
	-device virtio-net-pci,netdev=net \
	-netdev type=user,id=net,hostfwd=tcp::2222-:22 \
#	-netdev type=bridge,br=virbr0,helper=/usr/lib/qemu/qemu-bridge-helper,id=net  # for this see https://wiki.qemu.org/Features/HelperNetworking
#	-serial stdio
```