Ysurac/openmptcprouter #2007: Issues with virtualbox

cro888888, 2021-08-03:
I had each IP limited at 50Mbps, I got two network cards, whenever I activate two cards at same time when I go 192.168.0.99(lan) on browser it stucks on 192.168.0.99 not entering 192.168.0.99/cgi-bin/luci.

And is it possible to use it without VPS? Or maybe set it up locally? And is single physical card with multiple virtual cards possible? Is that what MacVLAN does?

Ysurac, 2021-08-03:
Please use the template.
Check that you use a bridge access and that you can ping lan ip.
VPS is needed to aggregate speed, without a VPS OpenMPTCProuter do some load balancing. The VPS must be somewhere where speed to internet is more than aggregated speed you can have, can't be local.
You often don't need macvlan in a vm, only create network card you need and bridge them. If you want to use macvlan on VM, set netword card as promiscuous.

cro888888, 2021-08-03:
Ping into VM works, the other way doesn't

closed 2021-11-06 by github-actions[bot]