Ysurac/openmptcprouter #2008: Add toggle/option to set TTL to 65?

Fffrank, 2021-08-04:
**Is your feature request related to a problem? Please describe.**
Many/most LET/4g/5g providers will detect tethering by inspecting packet TTL values (and then de-prioritize or slow them down.)

**Describe the solution you'd like**
Add toggle or option to automatically add firewall rule to mangle packet TTLs.
example: `iptables -t mangle -I POSTROUTING -o usb0 -j TTL --ttl-set 65`

**Describe alternatives you've considered**
Right now these commands can be put in the user-firewall but might lead to less speed complaints if it's done automatically.  No down-side?

**Additional context**
https://www.maroonmed.com/ttl-modification-for-outgoing-traffic-with-openwrt/ for reference

Ysurac, 2021-08-07:
Do you have examples of ISP and country still using this ?
It's difficult to test for me, did you have some good results with this ?

Fffrank, 2021-08-09:
Both T-Mobile and Verizon in the US use this method to detect tethering.  I see speeds go from 250-400kbps to 7-10mbps when over-riding the TTL.

Ysurac, 2021-08-09:
ok, I will try to find a good way to add it.

closed 2021-11-12 by github-actions[bot]