Ysurac/openmptcprouter #2010: 6in4 permission denied

zulu14, 2021-08-05:
## Expected Behavior
hello yannick, IPV6 6IN4 not working , but the interface are UP , the ipv4 tunnel are OK, proxy (SS) are OK 

## Current Behavior
ping 6in4-omr6in4 -> omr-6in4-user0: OK

ping omr-6in4-user0 -> 6in4-omr6in4: 100%packet loss or "invalid argument" if i not specifiy the interface in ping

ping  6in4-omr6in4 -> google ipv6 (ipv6 not domain) permission denied (user as root and try with sudo (seem to be a network denied)

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 1026
  - OpenMPTCProuter platform: X86

[image]
[image]
[image]
[image]

Ysurac, 2021-08-06:
IPv6 is enabled in wizard ?

zulu14, 2021-08-06:
hello yannick, yes the ipv6 is enabled, in syslog we have the omr-6in4 log say its enabled

zulu14, 2021-08-06:
have you already get the "permission denied" error when you dev the 6in4 system ?

Ysurac, 2021-08-06:
What tis the result of ```ip -6 a``` and ```ip -6 r``` on the router via SSH ?
Check also that omr-6in4 is in correct firewall zone (vpn zone)

zulu14, 2021-08-06:
6in4 are in VPN zone 
[image]

zulu14, 2021-08-06:
i have this in log: 
[image]

zulu14, 2021-08-06:
(lan0 is a mcvlan of eth1)

Ysurac, 2021-08-06:
There is no public/lan ipv6 on lan
check Network->Interfaces, LAN, DHCP Server tab, IPv6 settings

zulu14, 2021-08-06:
[image]

Ysurac, 2021-08-06:
Seems to be ok, what you have in Network->Interfaces, Global network options tab as ULA-Prefix ?

zulu14, 2021-08-06:
empty

zulu14, 2021-08-06:
[image]

Ysurac, 2021-08-06:
It's the problem.
You should set something like "fd53:8a41:3f89::/48"

zulu14, 2021-08-06:
okay, i have only 1 ipv6 from the vps,

i setup like this: 
[image]

if i ping test : 
[image]

zulu14, 2021-08-06:
(shorewall6 are OK and running )
ping ipv6.google.com on server side are OK

Ysurac, 2021-08-06:
Check that you get an IPv6 on lan interface. If not, reboot.

zulu14, 2021-08-06:
after reboot:

[image]

and same ping behavior

Ysurac, 2021-08-06:
What do you have in "ip -6 r" now ? What is the default ipv6 route ?

zulu14, 2021-08-06:
(tunnel: glorytun-TCP , proxy: SS , all wan are OK, proxy are OK, tunnel OK)

[image]

zulu14, 2021-08-06:
PING Test:
[image]

Ysurac, 2021-08-06:
Default IPv6 route seems to be missing

Ysurac, 2021-08-06:
Route should be added via ```/bin/omr-6in4``` script. Doesn't seems to work in your case. I will check why...

Ysurac, 2021-08-06:
What is the result of ```logread | grep omr6in``` ? (to check if script is sometimes running or not)
If empty, what is the result of ```/bin/omr-6in4``` ?

zulu14, 2021-08-06:
hello yannick , sorry for the delay, 
after checking, 6in4 bin file are empty !, i download it from the feeds (probably a missclick from me :/ ) 

the "permission denied" disapear and now its accept the ping, but he have no answer (100% packet loss)

zulu14, 2021-08-06:
he said in log (only one time): user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1

zulu14, 2021-08-07:
hello yannick little up

Ysurac, 2021-08-07:
I don't understand how you can have some empty file on your installation... What did you really do ?

zulu14, 2021-08-07:
we have build from feeds (fork) and i check the commit, i have make a mistakes and i have deleted this files , its now fixed
but i have 100% packet loss

Ysurac, 2021-08-07:
I don't support fork...
Check "ip r" and set default route.

closed 2021-08-09 by zulu14