Ysurac/openmptcprouter #2011: Can be used in read only?

highfreq, 2021-08-06:
If OMR is already configured and working, can it be used in its full potential if it goes in read only? or something is lost.
What i mean is, if the OMR system goes to read only AFTER the configuration is finished and has been working fine for some days.

Ysurac, 2021-08-06:
No, this will not work correctly.
But a FS should never be read only without a good reason like a power shortage (and with an UPS no problem).

highfreq, 2021-08-06:
System is on an appliance that is turned off just by taking off power and sometimes it happens that system goes in read only. It is a shame that OMR on raspberry can't be used with USB filesystem because it is a lot better and never goes in read only.

Ysurac, 2021-08-06:
It's not a good idea to do that. You can shutdown properly via the web interface.

highfreq, 2021-08-06:
I know but it is not a viable solution for us, appliance is on and off and we don't connect to it all the time with web browser.

Ysurac, 2021-08-06:
You can shutdown via SSH also ;)
RPI is really sensible to power shortage, I don't have any real possibility to fix that.

highfreq, 2021-08-07:
Maybe add support for USB boot/storage? USB storage is much more robust than SD.

closed 2021-11-10 by github-actions[bot]