Ysurac/openmptcprouter #2016: Max speed result on router hardware

Ysurac, 2021-08-07:
**Raspberry PI 4**: 240Mbps with chacha20 encryption
**Intel Celeron J1900**:

gsxryan, 2021-08-24:
I'll be benchmarking some soon as well with encryption.  I have 3x isp and total maximum 2190Mb down/ 1885Mb up.  I'll edit my results here when tested further.

**RPI3b+:** 
local RAW iperf3 limit: 150down/250up
no encryption 30 down / 15 up

**Raspberry PI 4 multiport (add: usb3.0 eth1):**
local RAW iperf3 limit: 120 / 250
no encryption 100 / 5

**Raspberry PI 4 single port macVLANs:**
local RAW iperf3 limit: 905/920
no encryption: 600/690

**Pending extended testing**
Intel 9700t - 2 CPU,  SHA256 GCM: 2.1Gb down / 1.8Gb up (and much more to go) $10 VPS
$5 [Linode](https://www.linode.com/?r=434ca22e2427430e5697563c488ab0d50920537d): 1200Mb Burst, levels at 1Gb/1Gb.

**Interested in testing:**
Protectli

closed 2021-11-28 by github-actions[bot]

ioogithub, 2022-09-03:
> **Raspberry PI 4 multiport (add: usb3.0 eth1):** local RAW iperf3 limit: 120 / 250 no encryption 100 / 5
> 
> **Raspberry PI 4 single port macVLANs:** local RAW iperf3 limit: 905/920 no encryption: 600/690

There is a huge discrepancy between your multiport and single port results. Do you attribute this to a bottleneck on the USB to LAN adapter or is there some other issue that causes this?

Ysurac, 2022-09-03:
This issue is old, and encryption speed is higher now (I get 325Mbps on same RPI4 with encryption)
USB on RPI is always quite bad, no idea if we still have such limit, some RPI firmware upgrade may have improved this.

ioogithub, 2022-09-03:
> I get 325Mbps on same RPI4 with encryption

Thanks Ysurac, this is very helpful. Just to clarify one point, are you getting 325Mbps with encryption using the Pi's onboard single Ethernet port or are you using Ethernet to USB dongles as  @Chris5342 mentioned in his Starlink discussion?

Ysurac, 2022-09-03:
Onboard ethernet port, I don't really like Ethernet to USB, there is often some problems.

ZZitrus, 2024-09-16:
Hey, the issue is no longer so current, but I couldn't find anything more suitable.
**Does anyone have test results for a Raspberry Pi 5?**
Buying a NanoPi R4S would be just as expensive as a Raspberry Pi 5 and the Raspi is much more powerful. I hope to be able to scratch the gigabit with it, as I would aggregate quite fast links.

I previously bought a Ubiquiti Edgerouter X just for this and got up to just under 100 Mbps (uplinks: 100 + 30 down), so I now know that hardware for OpenMPTC routers has to be really beefy. I may have done something wrong, but before that I thought it would have worked on mips24kc as well haha.