Ysurac/openmptcprouter #2017: Make a Pi installation more resilient?

hairy-dog, 2021-08-08:
I am running OMR on a Raspberry Pi4, and it works fairly well. But rather too often we have had power cuts. 

The Pi does not draw enough power to keep the UPS alive, so at present it is powered through a USB "powerbank" but still there have been power failures, and about half of the time this makes the SD card become Read Only. Sometimes it completely prevents the device booting.

It would be really good if there was some way to make the software installation on the Pi more able to withstand a power cut without being shut down properly.

Is there any way to do this?

kevin39, 2021-08-09:
Hi,
A powerbank to secure yourself from powercut doesn't seem the best solution...
Buy and setup an UPS. There are cheap model and it will allow you to keep your network UP in case of power-cut.

In my case I've a cheap UPS that keep my main switch & Rpi4 up in case of power-cut.
I also have 1 extra sdcard in case of SD failure.

hairy-dog, 2021-08-09:
If you had read my post, you would have noticed that I said it doesn't draw enough power to keep the UPS on. 
You are right: a power bank is not the best solution. The best solution is a system that doesn't corrupt it's file system if there is a power glitch.

yannismate, 2021-08-09:
You can usually prevent the filesystem on the SD card from corrupting on power loss by setting it up as read-only in the first place. If you don't plan on changing the configuration of your Pi/OMR too much that might be a good idea.
There's plenty of tutorials/ documentation online on how to set this up with 3rd party software running on the pie that needs to write temporary files.

If you plan on keeping it rw, you can only decrease the chance of the fs corrupting, but not eliminate it.

hairy-dog, 2021-08-10:
That's a very interesting idea. Looking at https://medium.com/swlh/make-your-raspberry-pi-file-system-read-only-raspbian-buster-c558694de79 it seems as if it is possible to make the SD filesystem read only, but to be able to toggle this with a shell script. 
If that could be incorporated into OMR so that when settings are being changed the filesystem is made RW but all the rest of the time it was RO that would possibly make the system more reliable. 
Is there a reason why that would not be a good idea?

ImprezyLive, 2021-08-10:
Add this to /etc/rc.local before exit 0
This procedure check your file system if is set to Read Only, repair and reboot PI.

dane=$(mount | grep '/dev/root' | awk '{print $6}')
if [ "$dane" != "(rw,noatime)" ]
then
    fsck.ext4 -y /dev/mmcblk0p2
        reboot
fi