Ysurac/openmptcprouter #2018: Interface not reconnected

bambuca, 2021-08-13:
## Expected Behavior

## Current Behavior
I have two interfaces, one is 300 Mb/s and second 100 Mb/s.
The first one is often disconnected  - "No Server ping response after 1 seconds". Speed is then 100 Mb/s.
When I restart the interface then the speed is again almost 400 Mb/s.
It would be nice if the interface is self restarted in that case.

[image]

Log:
```
Thu Aug 12 17:36:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 17:41:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 17:41:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 17:46:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 17:46:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 17:51:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 17:51:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 17:56:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 17:56:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:00:00 2021 cron.err crond[4348]: USER root pid 17286 cmd /etc/init.d/vnstat_backup backup
Thu Aug 12 18:01:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:01:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:06:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:06:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:08:50 2021 daemon.err /usr/bin/ss-redir[13774]: send: Connection reset by peer
Thu Aug 12 18:11:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:11:17 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:16:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:16:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:21:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:21:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:26:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:26:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:31:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:31:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:36:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:36:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:41:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:41:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:41:40 2021 daemon.info dnsmasq-dhcp[5460]: DHCPREQUEST(eth0) 192.168.100.100 e0:63:da:57:ce:eb
Thu Aug 12 18:41:40 2021 daemon.info dnsmasq-dhcp[5460]: DHCPACK(eth0) 192.168.100.100 e0:63:da:57:ce:eb dreammachine
Thu Aug 12 18:46:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:46:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:51:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:51:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 18:56:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 18:56:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 19:01:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 19:01:18 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.127455] usb 1-1.2: USB disconnect, device number 17
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.214733] usb 1-1.2: new high-speed USB device number 18 using xhci_hcd
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.336931] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.17
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.345470] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.352953] usb 1-1.2: Product: 4-Port USB 2.0 Hub
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.357915] usb 1-1.2: Manufacturer: Generic
Thu Aug 12 19:01:47 2021 kern.debug kernel: [1750883.362522] usb 1-1.2: no of_node; not parsing pinctrl DT
Thu Aug 12 19:01:47 2021 kern.debug kernel: [1750883.363789] hub 1-1.2:1.0: no of_node; not parsing pinctrl DT
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.363813] hub 1-1.2:1.0: USB hub found
Thu Aug 12 19:01:47 2021 kern.info kernel: [1750883.369074] hub 1-1.2:1.0: 4 ports detected
Thu Aug 12 19:06:16 2021 daemon.err /usr/bin/ss-redir[13774]: server recv: Host is unreachable
Thu Aug 12 19:06:16 2021 daemon.err /usr/bin/ss-redir[13774]: server recv: Host is unreachable
Thu Aug 12 19:06:19 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 19:06:19 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 19:09:06 2021 daemon.err /usr/bin/ss-redir[13774]: send: Connection reset by peer
Thu Aug 12 19:11:19 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Thu Aug 12 19:11:19 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.70 obtained, lease time 600
Thu Aug 12 19:12:54 2021 daemon.err /usr/bin/ss-redir[13772]: server recv: Host is unreachable
Thu Aug 12 19:16:19 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
[... 576 lines cut ...]
Fri Aug 13 08:56:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 08:56:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:01:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:01:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:06:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:06:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:11:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:11:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:16:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:16:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:20:23 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Aug 13 09:20:26 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Aug 13 09:21:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:21:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:26:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:26:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600
Fri Aug 13 09:28:32 2021 daemon.err /usr/bin/ss-redir[13772]: server recv: Host is unreachable
Fri Aug 13 09:31:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: sending renew to 10.122.129.1
Fri Aug 13 09:31:35 2021 daemon.notice netifd: wan1 (29930): udhcpc: lease of 10.122.129.71 obtained, lease time 600```

## Specifications

  - OpenMPTCProuter version:  v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform:  RPI3

Ysurac, 2021-08-13:
You can use services->omr-tracker to restart an interface if detected down (check available options)
But you seems to have an issue with your USB hub that go up and down. Check that it's a powered HUB, if yes use a better hub.

bambuca, 2021-08-19:
Thanks, I found the auto restart setting and it works. The USB hub is part of RPI, so I'll probably try to connect an external one with its own power supply.

closed 2021-08-19 by bambuca