Ysurac/openmptcprouter #2019: Cannot access at 192.168.100.1

keisuke-na, 2021-08-13:
I used RPi4B.
I tried installation on document but not work.
Cannot access at 192.168.100.1 orz

Ysurac, 2021-08-13:
Check that you get an IP via DHCP on your client computer in 192.168.100.x. If not set a static IP in this network (not 192.168.100.1 already used by the router).

keisuke-na, 2021-08-13:
RPi4B cannot get an 192.168.100.x. 
Should I install to VPS before access at 192.168.100.1?

Ysurac, 2021-08-13:
OpenMPTCProuter have a fixed 192.168.100.1 and it's also a DHCP server that give IP on 192.168.100.x network.
When you set it cannot get an 192.168.100.x IP, you attached a screen to check that ?

keisuke-na, 2021-08-13:
mmm...
Like this.
https://photos.app.goo.gl/9R4DKyvAmHtQBEKYA
What shoud I fix...

Ysurac, 2021-08-13:
Make sure you used correct image to flash the sd card, then connect a computer to the RPI via ethernet cable directly.
Then check IP of your computer.

keisuke-na, 2021-08-13:
oh...I misunderstood. 
I use correct image to flash the sd card.
But Can i ask that why dose RPi4B need ethernet?
I dont have ethernet... orz

Ysurac, 2021-08-13:
RPI4 only listen on ethernet port by default.
Default user is "root" with empty password.

keisuke-na, 2021-08-13:
Which mean MPTCProuter can't use the internet outside?
I want to try 4K streaming  from outside to Youtube with OBS.

Ysurac, 2021-08-13:
I don't understand what you mean. OpenMPTCProuter is a router, so it route traffic. RPI have only one internet port so it must be connected to an ethernet switch. Wifi can be used as client or AP, but RPI wifi is often really bad.

keisuke-na, 2021-08-13:
I have bad English. Im sorry.
Maybe I have misunderstood.
I've thought  that MPTCProuter has high speed features as internet connection when it has many 4G connection...

Ysurac, 2021-08-13:
Yes OpenMPTCProuter aggregate internet connections.

keisuke-na, 2021-08-13:
Can I use MPTCProuter outside as mobile hotspot??

Ysurac, 2021-08-13:
Yes, but for configuration at least you need an ethernet cable.

gsxryan, 2021-08-13:
RPI4 port will start as the default configuration port, plug your computer directly into the PI4 NIC.  You'll need a VLAN enabled switch to connect the PI4 to, and set your VLANS on the switch before enabling them on the RPI4. 

I plan to put together a video demonstrating how i set mine up soon.

FYI: Plan for these limits;
Wifi <200Mb
USBEth+PI4ETH: 300Mb
Eth+VLANs switch: 900Mb

keisuke-na, 2021-08-14:
@Ysurac 
mmm... 
I cannot use MPTCProuter without ethernet cable as mobile hotspot, right?

Ysurac, 2021-08-14:
@keisuke-na as I already said you can. But for first configuration you need to use a cable.

keisuke-na, 2021-08-14:
Ok. I understand.
Thanks for advice!

closed 2021-08-15 by keisuke-na