Ysurac/openmptcprouter #2020: Multipath seems to be blocked on the connection

h45h3ncryp73d, 2021-08-14:
## Expected Behavior
Everything should work fine? ;-)
Line bonding should work just fine.

## Current Behavior
Since I've upgraded (upgrade, not fresh install) OMR from 0.57.3 to 0.58.5 I get Multipath seems to be blocked on the connection on one of the LTE routers which had no such error previously. I guess something might have been blocked in the previous OMR version as well but maybe you just added a warning to make it visible in the new version. Anyway when I run speedtest.net I should get bonded speed but it's not the case. I only get connected via router which has been set as Master.
Both routers are the same but ISPs are different so I even tried switching SIM cards to make sure it's not the router thing, but the same error persisted.
I've seen other post regarding this error but with no solution so I've decided to open a new topic.

Also I've tried checking "MPTCP over VPN" option on affected router (only) but it's also not working - I get "No IP defined" and "Gateway DOWN" errors.

Only thing I can think of and try is to do a fresh install with factory image.

Any idea how to solve this issue? Thanks.

PS.
Just out of curiosity - is it safe to do installed software upgrades (System -> Software) or should they be avoided?

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026 5.4.65-mptcp
  - OpenMPTCProuter platform: RPi 4b x64

Ysurac, 2021-08-14:
Kernel with VPS 0.1026 is 5.4.100, make sure you rebooted after update.
If you have No IP defined or Gateway down then you have a problem on WAN configuration. MPTCP over VPN should only be used when MPTCP is blocked on a connection.

h45h3ncryp73d, 2021-08-14:
Will try rebooting VPS (can't do it right now as I currently cannot loose a connection) but I don't think that's an issue as I have rebooted a newly installed VPS (I've set it up just for testing purposes so I haven't posted its' version) and the same error occurred.
On the other hand (I'm no network expert) but I haven't changed any WAN settings prior nor after OMR update. Since I've tried switching SIM cards between routers and I only have this issue with one ISP regardless of the router used I guess this could be ISP related (blocking) but still why wasn't there any problems prior to update?
Do you have any recommendation if I can do any test(s) to diagnose the problem in detail?
Thanks.

h45h3ncryp73d, 2021-08-14:
Do you recommend updating software packages manually or should they be left as they are due to some OMR specifics? I've updated some of them but I did it before too without any problems ...

Ysurac, 2021-08-14:
You shouldn't update packages.
I don't know what error you mean. Gateway or no ip defined are only related on router configuration. Check config or do a fresh configuration. Check also State->System log. If you use ethernet router/modem you shouldn't have any problems.

h45h3ncryp73d, 2021-08-14:
Problem is definitely not related to routers.. I'll try flashing a clean installation and see if it will solve "Multipath seems to be blocked ..." problem. It is definitely blocking proper OMR function.

Ysurac, 2021-08-14:
The "Multipath seems to be blocked..." is only a warning. You should test in network->MPTCP if it's really blocked or not. Some CGNAT are detected as blocked even if it's not the case.

h45h3ncryp73d, 2021-08-14:
> The "Multipath seems to be blocked..." is only a warning. You should test in network->MPTCP if it's really blocked or not. Some CGNAT are detected as blocked even if it's not the case.

Yes it seems MPTCP to be blocked on one of the ISPs as I got "-TCPOptionMPTCPCapable". Could you tell me in which OMR version have you implemented this warning? Was it present in 0.57.3? I guess this has nothing to do with OMR rather with ISP but still just wanted to rule out every single possibility that new OMR version has something to do with it.
As I've said before I'm not that great networking expert but could it be possible for ISP to block my connection if I use "MPTCP over VPN" option as it is also not working for me (as mentioned in the first post).

Ysurac, 2021-08-15:
It's available since a long time but I changed method to detect that in 0.58. It's only detected by status page, nothing else use it.
You should use MPTCP over VPN only on blocked WAN. It's a workaround, not really ideal. You can choose between Wireguard or OpenVPN for MPTCP over VPN in wizard advanced settings checkbox.

h45h3ncryp73d, 2021-08-15:
Thank you for your help. WireGuard is not working for me, will try OpenVPN.
Also I did some research and seems like affected ISP does use CGNAT ... I've called them and requested to remove it so I could use a port forwarding and they said they've removed it, but it's still blocked. Will try few more thing and calling again.
Is there actually a way for us ordinary users to detect CGNAT in use? Except of course line bonding is not working in OMR? From what I've read about it I actually cannot detect it in OMR Status page as I'm not seeing any local IP addresses on wan interface on router with affected ISP, but I shouldn't be seeing any anyway, right? I want to know because I really want to push my ISP to make it go away but I need solid proof.

Ysurac, 2021-08-15:
You should have in any case a local IP on wan interface. It's set as DHCP or static IP ?

h45h3ncryp73d, 2021-08-15:
> You should have in any case a local IP on wan interface. It's set as DHCP or static IP ?

Funny thing is I don't have local IP address under WAN interface on multipath blocked router. I have "regular/external" IP address searchable/available in WHOIS database (registered to my ISP provider).
Regarding DHCP/static IP ... since I'm not quite sure what you're referring to I'll give you answer to all possibilities:
1) My ISP assigns dynamic IP addresses - I haven't really watched over this but I guess they change daily or upon reconnect;
2) My router is set to use DHCP for OMR, therefore
3) OMR is using DHCP from my router to assign IP (although DHCP IP address it actually rarely, if ever changes).

h45h3ncryp73d, 2021-08-20:
Small update: I did a fresh install and at least for now MPTCP over VPN option is working for connection with blocked multipath. At least I think it is working as I can see a speedtest bandwidth increase and MPTCP tests are successful on both WAN interfaces.
But now I do have another problem ... after reboot I regularly get Filesystem is read-only ...  
Tried clean install with two SD cards - SanDisk Extreme Pro 128GB and Samsung Pro 64GB. And I have official RPi 4 power adapter.
And for some reason the OMR icon on the status page is at first green and the name of my router which is issuing IP address to OMR is shown but afterwards I just get "Your IP address was not leased by this router". Does this have to do something with MPTCP over VPN option being enables (I guess it should not be related) ...

Ysurac, 2021-08-20:
Do you reboot using interface or by removing power ? If you remove power this can corrupt filesystem.
The "Your IP address was not leased by this router" happen if you use static IP or if computer doesn't request an IP yet. It's only some info.

h45h3ncryp73d, 2021-08-20:
Well I got a power outage which surpassed time my UPS could handle so the first time shutdown was a "bit" forceful and I get it could cause filesystem to get corrupted. But second time I did a nice reboot from cli ...
Yes my router is set to provide OMR with static IP ...
For now I won't reboot the system as I need Internet access but will test it out asap.

And please if you couId give me some (offtopic) advice - If I have a router before OMR, and router after OMR which is providing WAN access and I want to optimize let's say gaming ping on which router should I do it on pre-OMR, OMR, post-OMR or on all of them? I guess the right answer would be post-OMR router providing WAN access, but still I just want to be sure without getting  into too much networking.
And another question related to custom DNS servers ... I'm using NextDNS.io and I also guess I need to set it on both OMR and post-OMR router to be sure all requests are using NextDNS, right?
Yeah, noob questions which will save me at least few hours of exploring, if you could answer them ...

Anyway, thank you.

Ysurac, 2021-08-20:
What you mean by optimize gaming ping ?
Custom DNS must be set in OpenMPTCProuter.

h45h3ncryp73d, 2021-08-20:
> What you mean by optimize gaming ping ?

I've seen some YouTube videos where MikroTik LTE routers can be optimized for lower ping ... so I'm just curious if OMR has such capabilities or is it just specific to MikroTik.

h45h3ncryp73d, 2021-08-20:
> Custom DNS must be set in OpenMPTCProuter.

I've installed NextDNS package (maybe the wrong approach but somehow seemed the easiest way to do) but when I try to enable NextDNS it cannot bind with port 53 as it is being used. The only place I could find mention of port 53 is under Network -> DHCP and DNS -> Advanced Settings -> DNS server port. How can I make NextDNS use port 53? I really don't want to make it work without messing up my Internet connection ...

Ysurac, 2021-08-20:
It's not the good approach (or make it listen to another port than 53 if possible).
Custom DNS must be set in Network->DHCP&DNS, in "DNS forwardings" replace "127.0.0.1#5353" by the DNS you want.
If you use another DNS server than unbound, at least omr-bypass will not work.

h45h3ncryp73d, 2021-08-20:
NextDNS has the following option for unbound:

Use the following in unbound.conf:
forward-zone:
  name: "."
  forward-tls-upstream: yes
  forward-addr: 45.90.28.0#b2e21f.dns1.nextdns.io
  forward-addr: 2a07:a8c0::#b2e21f.dns1.nextdns.io
  forward-addr: 45.90.30.0#b2e21f.dns2.nextdns.io
  forward-addr: 2a07:a8c1::#b2e21f.dns2.nextdns.io
As a recursive resolver, Unbound chases CNAMEs. This may result in unexpected behavior when used in conjunction with a blocking DNS resolver like NextDNS. See github.com/NLnetLabs/unbound/issues/132.

Maybe I'm boring you but would you recommend this approach regarding OMR services or not?

h45h3ncryp73d, 2021-08-20:
> ... in Network->DHCP&DNS, in "DNS forwardings" replace "[127.0.0.1#5353](//127.0.0.1#5353)" by the DNS you want.

Can I replace it by  domain name like xyz.dns.nextdns.io without port?

Ysurac, 2021-08-20:
yes you can replace it with "xyz.nextdns.io", the #port is only needed when port is not the default one: 53

h45h3ncryp73d, 2021-08-20:
> yes you can replace it with "[xyz.nextdns.io](//xyz.nextdns.io)", the #port is only needed when port is not the default one: 53

Thanks. Should it be the first one or not? I guess not since under Advanced Settings -> Strict order is not checked ...

h45h3ncryp73d, 2021-08-20:
Maybe I'm crazy but since I've changed DNS I've got error on Status page under VPS "No server IP address, No public IP address Can't get public IP address from ShadowSocks" ... but when I check my IP it's showing my VPS IP address ...

Ysurac, 2021-08-20:
It's because it can't contact the site used to get public ip ( ip.openmptcprouter.com ) and others way based on DNS doesn't work too.

h45h3ncryp73d, 2021-08-20:
> It's because it can't contact the site used to get public ip ( [ip.openmptcprouter.com](//ip.openmptcprouter.com) ) and others way based on DNS doesn't work too.

Sorry I'm confused now ... obviously my knowledge is limiting me to obtain the "full picture" ... Is it working or not?
If I go to https://ip.openmptcprouter.com/ in my browser I can see my VPS (public) IP address, so how come OMR cannot retreive it?
And I can access the Internet through VPS even this error is shown ...

Ysurac, 2021-08-20:
Can you try on the router: ```curl ip.openmptcprouter.com``` via SSH ?
Check also if you have errors in State->System log

h45h3ncryp73d, 2021-08-20:
> curl [ip.openmptcprouter.com](//ip.openmptcprouter.com)

Not working.

Hmm, here is log (sanitized) and fishy dnsmasq errors from it:

Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev tun0 ingress
Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev tun0 root
Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev ifb4tun0 root
Fri Aug 20 19:08:21 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Fri Aug 20 19:08:21 2021 user.notice SQM: Starting SQM script: piece_of_cake.qos on tun0, in: 0 Kbps, out: 0 Kbps
Fri Aug 20 19:08:22 2021 user.notice SQM: piece_of_cake.qos was started on tun0 successfully
Fri Aug 20 19:08:22 2021 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Fri Aug 20 19:08:22 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Fri Aug 20 19:08:22 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Aug 20 19:08:22 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Fri Aug 20 19:08:22 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Fri Aug 20 19:08:23 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http 1.0.0.1)
Fri Aug 20 19:08:24 2021 daemon.err glorytun[28052]: <VPS_IP>.65001: connected
Fri Aug 20 19:08:24 2021 daemon.info glorytun[28052]: STARTED tun0
Fri Aug 20 19:08:24 2021 user.notice omr-bypass: Restart dnsmasq...
Fri Aug 20 19:08:24 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Fri Aug 20 19:08:24 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Aug 20 19:08:24 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Aug 20 19:08:24 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Fri Aug 20 19:08:24 2021 daemon.crit dnsmasq[30888]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:24 2021 daemon.crit dnsmasq[30888]: FAILED to start up
Fri Aug 20 19:08:24 2021 user.notice omr-bypass: OMR-ByPass is running
Fri Aug 20 19:08:24 2021 user.notice OMR-VPS: Can't get <VPS_NAME> token, try later (can't ping server <VPS_NAME> on <VPS_IP>, server API answer on <VPS_IP>)
Fri Aug 20 19:08:29 2021 daemon.crit dnsmasq[32246]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:29 2021 daemon.crit dnsmasq[32246]: FAILED to start up
Fri Aug 20 19:08:30 2021 user.notice omr-tracker: Launched
Fri Aug 20 19:08:30 2021 daemon.info vnstatd[14784]: Info: SIGTERM received, exiting.
Fri Aug 20 19:08:30 2021 daemon.info vnstatd[32602]: Info: vnStat daemon 2.6 started. (pid:32602 uid:0 gid:0 64-bit)
Fri Aug 20 19:08:30 2021 daemon.info vnstatd[32602]: Info: Monitoring (2): eth2 (1000 Mbit) eth1 (1000 Mbit)
Fri Aug 20 19:08:34 2021 daemon.crit dnsmasq[851]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:34 2021 daemon.crit dnsmasq[851]: FAILED to start up
Fri Aug 20 19:08:39 2021 daemon.crit dnsmasq[2139]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:39 2021 daemon.crit dnsmasq[2139]: FAILED to start up
Fri Aug 20 19:08:44 2021 daemon.crit dnsmasq[2696]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:44 2021 daemon.crit dnsmasq[2696]: FAILED to start up
Fri Aug 20 19:08:45 2021 user.notice post-tracking-post-tracking: Set firewall on server
Fri Aug 20 19:08:46 2021 user.notice omr-bypass: Starting OMR-ByPass...
Fri Aug 20 19:08:49 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Fri Aug 20 19:08:49 2021 daemon.crit dnsmasq[4884]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:49 2021 daemon.crit dnsmasq[4884]: FAILED to start up
Fri Aug 20 19:08:49 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Fri Aug 20 19:08:51 2021 user.notice omr-bypass: Reload dnsmasq...
Fri Aug 20 19:08:51 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Aug 20 19:08:51 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Aug 20 19:08:51 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Fri Aug 20 19:08:51 2021 daemon.crit dnsmasq[6280]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:08:51 2021 daemon.crit dnsmasq[6280]: FAILED to start up
Fri Aug 20 19:08:51 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 7 crashes, 0 seconds since last crash
Fri Aug 20 19:08:51 2021 user.notice omr-bypass: OMR-ByPass is running
Fri Aug 20 19:08:55 2021 user.notice omr-bypass: Starting OMR-ByPass...
Fri Aug 20 19:08:56 2021 user.notice omr-bypass: Restart dnsmasq...
Fri Aug 20 19:08:56 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Aug 20 19:08:56 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
[... cut ...]
Fri Aug 20 19:09:58 2021 daemon.crit dnsmasq[21083]: FAILED to start up
Fri Aug 20 19:10:03 2021 daemon.crit dnsmasq[21928]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:10:03 2021 daemon.crit dnsmasq[21928]: FAILED to start up
Fri Aug 20 19:10:08 2021 daemon.crit dnsmasq[22763]: bad address at line 15 of /var/etc/dnsmasq.conf.cfg01411c
Fri Aug 20 19:10:08 2021 daemon.crit dnsmasq[22763]: FAILED to start up
Fri Aug 20 19:10:08 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

Ysurac, 2021-08-20:
#1901 was answered: only shadowsocks is supported with MPTCP over VPN.
You issue is easy: you make a mistake in your change, so dnsmasq can't start.

closed 2021-11-24 by github-actions[bot]

talitha77, 2024-09-16:
using 5.9 with my older vps and all ok.. three starlinks green and happy.
I have a new rpi5 with latest OMR and new vps with  6.1.0-22-amd64 # Debian 6.1.94-1 (2024-06-21) x86_64 GNU/Linux
and now all three starlinks say that mptcp appears to be blocked. If i swap to the older vps that message goes away on one or two of the wans (of course i get a kernel mismatch)..  
[image]
[image]

talitha77, 2024-09-16:
Sep 16 07:07:28 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 78:0c:f0:21:f5:74 
Sep 16 07:07:28 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.101.212 78:0c:f0:21:f5:74 
Sep 16 07:07:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.101.212 78:0c:f0:21:f5:74 
Sep 16 07:07:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.101.212 78:0c:f0:21:f5:74 WANMeister2
Sep 16 07:09:00 OpenMPTCProuter daemon.info vnstatd[11415]: Info: Traffic rate for "eth4" higher than set maximum 10 Mbit (20s->27262976, r87372314 t4311607, 64bit:1), syncing.
Sep 16 07:09:20 OpenMPTCProuter daemon.info vnstatd[11415]: Info: Detected bandwidth limit for "eth4" changed from 10 Mbit to 1000 Mbit.
Sep 16 07:09:20 OpenMPTCProuter daemon.info vnstatd[11415]: Info: Detected bandwidth limit for "eth3" changed from 10 Mbit to 1000 Mbit.
Sep 16 07:18:40 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 10.2.0.154 error (114.114.115.115,1.2.4.8,80.67.169.40)
Sep 16 07:18:40 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 185.181.9.224 dev eth2
Sep 16 07:18:40 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 185.127.19.147 dev eth2
Sep 16 07:18:47 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched up
Sep 16 07:18:47 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Interface route not yet set, set route ip r add default via 10.2.0.1 dev eth2 metric 8
Sep 16 07:19:10 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 10.2.0.154 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sep 16 07:19:10 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 185.181.9.224 dev eth2
Sep 16 07:19:10 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 185.127.19.147 dev eth2
Sep 16 07:21:23 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: remote recv: Connection reset by peer
Sep 16 07:22:28 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched up
Sep 16 07:22:28 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Interface route not yet set, set route ip r add default via 10.2.0.1 dev eth2 metric 8
Sep 16 07:22:30 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for wan2 (eth2): 216.128.25.204 (previous: 216.128.25.41)
Sep 16 07:22:30 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for eth2
Sep 16 07:23:45 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: remote recv: Socket not connected
Sep 16 07:26:36 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: send: Connection reset by peer
Sep 16 07:26:36 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: send: Connection reset by peer
Sep 16 07:26:57 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:26:57 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:27:02 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:27:18 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: server recv: Operation timed out
Sep 16 07:27:23 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:27:43 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: server recv: Operation timed out
Sep 16 07:27:54 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:27:54 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9008]: server recv: Operation timed out
Sep 16 07:28:04 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: server recv: Operation timed out
Sep 16 07:28:21 OpenMPTCProuter user.notice MPTCP: Set eth0 to on from eth0 is deactivated
Sep 16 07:28:21 OpenMPTCProuter daemon.notice ttyd[716]: [2024/09/16 07:28:21:7031] N: rops_handle_POLLIN_netlink: DELADDR
Sep 16 07:28:24 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 16 07:28:24 OpenMPTCProuter user.notice omr-tracker: Launching...
Sep 16 07:28:24 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: server recv: Operation timed out
Sep 16 07:28:29 OpenMPTCProuter daemon.err /usr/bin/ss-redir[9009]: server recv: Operation timed out
Sep 16 07:28:35 OpenMPTCProuter user.notice omr-tracker: Launched
Sep 16 07:28:39 OpenMPTCProuter user.notice omr-tracker: Launching...
Sep 16 07:28:50 OpenMPTCProuter user.notice omr-tracker: Launched
Sep 16 07:28:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sep 16 07:28:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Sep 16 07:28:58 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sep 16 07:29:10 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Sep 16 07:29:10 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Sep 16 07:29:11 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 16 07:29:11 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
[... cut ...]
Sep 16 07:45:18 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Sep 16 07:45:18 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sep 16 07:45:30 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Sep 16 07:45:30 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Sep 16 07:45:31 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 16 07:45:31 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

Ysurac, 2024-09-16:
Please open a new issue.