Ysurac/openmptcprouter #2021: Questions about IPv6 internet access on IPv4-only connections with IPv6 VPS

milankragujevic, 2021-08-15:
## Expected Behavior
Given 3 IPv4-only connections for OMR (LTE+WISP+DSL), and a VPS with IPv6 network (Hetzner), I want to be able to reach IPv6 services on my local network (OMR).

## Current Behavior
When configuring IPv6, the Wizard requires additional IPs to be added for each WAN interface. I am not sure what is the purpose of this.

My requirement is **just** to get IPv6 access working, for educational purposes. The WAN connections are V4-only. I expect that V6 can be routed inside the tunnels and/or with the proxy, over V4-only WAN. 

## Specifications

  - OpenMPTCProuter version: v0.58.5
  - OpenMPTCProuter VPS version: v0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-08-17:
The wizard doesn't need additional IPs to be added on each WAN, but WAN IPv6 can be configured if available. Else VPN tunnel is used for IPv6 over IPv4.
By default a private IPv6 is distributed to clients, if your provider give a large IPv6 subnet you can also use it and IPv6 will be available from outside (if allowed in firewall).

milankragujevic, 2021-08-17:
But it shows fields in wan configuration and marks them red if not entered. I assumed it was required.

Adorfer, 2021-10-12:
to anser the question: 
>My requirement is just to get IPv6 access working, for educational purposes. The WAN connections are V4-only. I expect that V6 can be routed inside the tunnels and/or with the proxy, over V4-only WAN.

i assume that the requestest functionality is: 
"inbound connections from WORLD to IPv6 on the OMR LAN"

in this case the answer is: 
the function is NOT depending upon ipv4 or ipv6 present or absence on any WAN insterface, as long as at least one is connected to the VPS via one of the address families.

closed 2022-01-15 by github-actions[bot]