Ysurac/openmptcprouter #2022: Second LAN for modems

mycanaletto, 2021-08-15:
Hello,

I added a second ethernet card (eth1) on a second subnet (192.168.110.2/24) configured in static in LAN zone on firewall side. (openMPTCP in VM).

From this subnet I ping well this interface.

However, from MPTCP (ssh console) it is impossible to ping the modems on this subnet and therefore to configure them.

Ysurac, 2021-08-17:
I don't understand, you say it's a LAN, why does it have modem on it ?
Check route and firewall if you can't ping a network.

mycanaletto, 2021-08-17:
There were always modems on LAN (the only network card in openmptcp). I just added one on the other VLAN port to join another box.

Ysurac, 2021-08-17:
You use MacVLAN or VLAN ?
You checked firewall zones ?

mycanaletto, 2021-08-17:
I tested all possibilities and both network cards are in the LAN on the FW side

Ysurac, 2021-08-17:
Please describe correctly your installation if you want some help. Make a schema or something like that.
For now I don't know what you have and what you want.

mycanaletto, 2021-08-17:
Note that on another Linux I added a network card in the same way and everything is ok

2 network cards :
- LAN1 192.168.210.0/24 : LAN clients + 1 modem : ok
- LAN2 192.168.110.0/24 : LAN clients + 2 modems

[image]

Ysurac, 2021-08-17:
This will not work really well this way.
LAN part is on same subnet as modems and multiple modems are on same subnet.
Check also wan2 and wan3 configuration (seems to be macvlan), they must be in WAN firewall zone.
VPS wasn't rebooted after update.

mycanaletto, 2021-08-17:
It doesn't explain why in console mode I don't ping the lan2 and that the reverse is possible. Special fw rule ?

[image]

[image]

[image]

mycanaletto, 2021-08-17:
In fact it works by itself, but under ESX, and as for the first LAN you must not forget to accept the Promiscuous mode...

[image]

closed 2021-11-21 by github-actions[bot]