Ysurac/openmptcprouter #2024: Booting raspberry pi 4 from usb (connected to ssd)

hi10bha, 2021-08-16:
I am able to boot with the connected usb ssd to raspbian os without any issues, but I m not able to boot in OMR can someone throw some light on this , it starts to boot but does not come to the prompt nor m able to login . It halts at attached SCSI removable disk.  I am looking to use Argon one V2 housing if this works for me

highfreq, 2021-08-17:
Asked the same 2 weeks ago, i would like to boot from usb stick, much more robust against readonly filesystem and much daster but i got no answer, if you find a solution please let me know.
I have OMR installed on an appliance that needs to be powered off not gracefully and i have to change SD once a week because of bad sectors.

hi10bha, 2021-08-17:
Exactly the same reason I want to move to ssd

Ysurac, 2021-08-17:
Please use templates...
You may need to update RPI firmware and modify /boot/cmdline.txt

highfreq, 2021-08-17:
*UPDATE*
Updated bootloader this morning on rpi4 to latest and changed to sda2 in cmdline.txt and now everything is working.

Ysurac, is it possible on an Rpi4 to attach on the same ethernet port (the only one) , 2 lan gateways and local lan?
I tried following your how to and giving just ip and gateway ip to wan 1 and wan 2 in tutorial but it doesn't work, interface down.
Should i use vlan?

Ysurac, 2021-08-17:
On RPI, by default MacVLAN is used. Make sure you doesn't change this (macvlan must be based on eth0 physical interface), then only IP and gateway settings are needed.

highfreq, 2021-08-17:
So i have to select macvlan?

Ysurac, 2021-08-17:
if you don't have any VLAN, yes, use MacVLAN

highfreq, 2021-08-17:
I need to have 2 modems and regular lan on the same ethernet port, so i leave lan as it is and configure the 2 wans as macvlan?

highfreq, 2021-08-17:
macvlans based on eth0 i guess?

highfreq, 2021-08-17:
I made macvlans and it works, untill i reboot than the wan interface says Unknown error (DEVICE_CLAIM_FAILED). Maybe it is referred to eth0 wich i am using it in bridge with wlan0. Should i remove the bridge?

hi10bha, 2021-08-17:
Where can I find cmdline.txt in openmptcp /boot

hi10bha, 2021-08-18:
thanks for all the help Ysurac and highfreq thankyou very much I managed to get my ssd running

jphgardner, 2021-10-09:
How did you get it working exactly?

closed 2022-01-12 by github-actions[bot]