Ysurac/openmptcprouter #2025: German ISP (M-Net) is blocking MPTCP - alternate methods not working

dennydeutscher, 2021-08-16:
Hi,

I had a problem on a particular internet connection with the ISP "MNet". If I use that connection the speed is dropping down to 
5 MBit/s Down and a few KBit/s Up. It seems to me the provider is blocking the mptcp traffic. 

The connection is a FTTB Solution where the last few meter in the building is implemented in VDSL.
The connection handles 99 MBit/s Download and 40 Mbit/s Upload before the MPTCP Router.

I tried also a connection from "deutsche Telekom" which is working perfectly. 

My question:

What can I provide you to figure out why the traffic is so slow and how to work around that?

Thank you, Denny

Adorfer, 2021-08-17:
>  If I use that connection the speed is dropping down to
> 5 MBit/s Down and a few KBit/s Up. It seems to me the provider is blocking the mptcp traffic.

sounds more like a throttle, not a block 

> alternate methods not working
> how to work around that?

If alternative methods do not work, then there are not alternative methods.

dennydeutscher, 2021-08-17:
> If alternative methods do not work, then there are not alternative methods.

even the openvpn tunnel? That seems totally strange to me... how will they detect such traffic?

Another weird thing: 

As soon a I set up mptcp the connection gets slow on that particular connection... BUT it's still slow even when I disable MPTCP in settings. Is there no way to set openmptcp router to act a a normal openwrt device after initial mptcp setup in case there is a problematic ISP?

Ysurac, 2021-08-17:
Check result of Network->MPTCP, "MPTCP support check" tab.

dennydeutscher, 2021-08-25:
Hey there. Good news... the speed limitation didn't came from the ISP. The bintec be.ip plus router which is installed as main gateway had an issue when Ethernet connection was set in "bridge-mode" instead of "routing-mode".

closed 2021-08-25 by dennydeutscher