Ysurac/openmptcprouter #2026: How to boot Rpi4 from USB stick Tutorial

highfreq, 2021-08-17:
1 - Download and burn proper Rpi4 image
2 - In Boot edit cmdline.txt change the root device from mmcblk0p2 to sda2
3 - Exit and save

Now your Rpi4 should boot from USB, if it doesn't make a raspian SD and boot into the OS. From command line issue following: 

1 - **sudo rpi-eeprom-update** to see if there is a new eeprom version
2 - **sudo rpi-eeprom-sudo rpi-eeprom-update -d -a** to update

At the end of the update it'll ask to reboot, do it.

After reboot done power off and take SD out and put your OMR USB stick in and enjoy.

Hope it is clear :)

hi10bha, 2021-08-17:
[image]

it gets stuck at the above point in image and n not able to run your second command 
2 - sudo rpi-eeprom-sudo rpi-eeprom-update -d -a to update

hi10bha, 2021-08-18:
thanks a lot, I managed. thanks

closed 2021-11-28 by github-actions[bot]