Ysurac/openmptcprouter #2029: omr-service[467]: jq: error (at <stdin>:1): Cannot iterate over null (null)

wlatic, 2021-08-17:
## Expected Behavior

Stability

## Current Behavior

Some Instability on connections + syslog contains:
omr-service[467]: jq: error (at <stdin>:1): Cannot iterate over null (null)

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: VULTR
  - OpenMPTCProuter platform:  x86_64

Want to confirm if omr-service[467]: jq: error (at <stdin>:1): Cannot iterate over null (null) is an error I should be looking at, its a completely fresh VPS install (today).

Ysurac, 2021-08-17:
Check your /etc/openmptcprouter-vps-admin/omr-admin-config.json file. It seems that there is a problem in it.

wlatic, 2021-08-17:
Looks fine:

```
{
    "port": 65500,
    "users": [
        {
            "admin": {
                "username": "admin",
                "user_password": "**",
                "permissions": "admin",
                "disabled": false
            },
            "openmptcprouter": {
                "username": "openmptcprouter",
                "user_password": "**",
                "shadowsocks_port": 65101,
                "disabled": false,
                "v2ray": {
                    "key": "**",
                    "port": "65228"
                },
                "vpnremoteip": "10.255.255.2",
                "vpnlocalip": "10.255.255.1"
            }
        }
    ],
    "debug": false,
    "allips": [
        "**"
    ],
    "ipv4": "**",
    "hostname": "**"
}
```

Ysurac, 2021-08-17:
ok. I fixed the script, it's because a value is not yet initialized (LAN ip). So this will be fixed when CDN update his cache.
It's not related to problems you can have elsewhere.

wlatic, 2021-08-17:
So wait for a while and then update?

Ysurac, 2021-08-17:
As it's not related and doesn't change anything, you don't need to update the script.
This will also be fixed when router contact the VPS API.

wlatic, 2021-08-18:
Ended up doing a full new version as was concerned that could be an issue.

Previously you've said having two connections going to the same place can be problematic. I've currently got two LTE connections connected to the same tower and provider, could this cause issues?

closed 2021-11-22 by github-actions[bot]