Ysurac/openmptcprouter #2030: new Mac OS bigger and Monterey versions issues

hi10bha, 2021-08-18:
## Expected Behavior

Am not able to surf internet on Macs running bigsur and Monterey, I can access internet on Catalina os through mptcp
I changed the glorytun to udp from tcp but still facing issues

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

finnschi, 2021-08-20:
i run bigsur and have no issues surfing? you need to do more troubleshooting  usually its a dhcp or dns issue

hi10bha, 2021-08-20:
Thanks for the reply

I am able to access the router and it's set to dhcp and dns is set to 8.8.8.8 still am not able to access

Ysurac, 2021-08-20:
You shouldn't modify DNS on OpenMPTCProuter.
Check:
* that you can ping OpenMPTCProuter IP from your clients. 
* if yes, check that you can access http://1.1.1.1/
* If yes, check a ```dig google.com``` result (if dig is available on macos)

hi10bha, 2021-08-21:
that you can ping OpenMPTCProuter IP from your clients. ( yes ) I can log in to 192.168.100.1

if yes, check that you can access http://1.1.1.1/ (NO)
If yes, check a dig google.com result (if dig is available on macos)

Ysurac, 2021-08-21:
So the problem is not related to DNS if you can go to http://1.1.1.1
Check status page, all should be green with VPS public IP displayed in VPS box.
If it's ok, check that you use OpenMPTCProuter IP as gateway. and that you can ```ping 1.1.1.1```

hi10bha, 2021-08-22:
in Status box

Status box is all green , everything runs fine on my windows laptop, no issues at all. Mac with Catalina - am able to run when I switch glory tun to udp. but in BIG SUR and Monterey NO.

closed 2021-09-07 by hi10bha