Ysurac/openmptcprouter #2031: Loggin in to Router after configuration

eothein, 2021-08-18:
## Expected Behavior

Configured the Router and the modems (MikroTik Routers):
- For WAN1: Protocol - Static address, IPv4 address - 192.168.10.2, Subnet mask - 255.255.255.0, IPv4 Gateway - 192.168.10.1
- For WAN2: Protocol - Static address, IPv4 address - 192.168.11.2, Subnet mask - 255.255.255.0, IPv4 Gateway - 192.168.11.1

Switched my laptop in the switch connected to both the router and the MikroTek's. 

**To be expected**: by surfing to 192.168.10.1 and 192.168.11.1, I should be able to log in to my Mikotek routers. 

*I need to login to the Mikrotik routers to perform some traffic shaping* 

## Current Behavior

Login fails.

If I change my IP to one of the domains (e.g. 192.168.10.100), I can log in to the Mikroteks. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform: RPI3

EmFl, 2021-08-25:
What's your current laptop setting ? Is your gateway correctly set to the router IP ?
When you say login fails, do you mean that you cannot reach the login page ?

eothein, 2021-09-02:
I managed to solve the issue: I was not on the same network, so manually added the 192.168.10.0/24 and the 192.168.11.0/24 network on my mac.

closed 2021-09-02 by eothein