Ysurac/openmptcprouter #2032: Basic setup does not work

eothein, 2021-08-18:
## Expected Behavior
Should have an internet connection up and running. 

Configuration: 
[image]

- Routers have a public IP. After login in to the router `ping 1.1.1.1` works. 
- Routers are able to ping 206.189.3.13, the VPS Server
- From **192.168.2.1** I can ping **192.168.10.2**

## Current Behavior

 Reference System log: 

````
Wed Aug 18 09:56:00 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Aug 18 09:56:10 2021 daemon.debug dnsmasq[23984]: listening on tun0(#178): 10.255.255.2 port 53
Wed Aug 18 09:56:10 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed Aug 18 09:56:11 2021 user.notice omr-tracker: Launching...
Wed Aug 18 09:56:18 2021 user.notice omr-tracker: Launched
Wed Aug 18 09:56:18 2021 daemon.debug dnsmasq[23984]: stopped listening on wan1(#170): fe80::d81d:5bff:fe78:ed99%wan1 port 53
Wed Aug 18 09:56:21 2021 daemon.err /usr/bin/ss-local[29246]: getpeername: Socket not connected
Wed Aug 18 09:56:36 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because check error and ping from 192.168.11.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Wed Aug 18 09:56:36 2021 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because check error and ping from 192.168.10.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Wed Aug 18 09:56:36 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Wed Aug 18 09:56:36 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev wan1
Wed Aug 18 09:56:41 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Aug 18 09:56:42 2021 daemon.err /usr/bin/ss-local[29246]: getpeername: Socket not connected
Wed Aug 18 09:56:43 2021 user.notice omr-bypass: Restart dnsmasq...
Wed Aug 18 09:56:43 2021 daemon.info dnsmasq[23984]: exiting on receipt of SIGTERM
Wed Aug 18 09:56:44 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Aug 18 09:56:44 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Aug 18 09:56:44 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: started, version 2.84 cachesize 150
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: DNS service limited to local subnets
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: DNSSEC validation enabled
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: DNSSEC signature timestamps not checked until receipt of SIGINT
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: configured with trust anchor for <root> keytag 20326
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq-dhcp[3538]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain test
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain onion
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain localhost
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain local
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain invalid
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain bind
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain use-application-dns.net
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using only locally-known addresses for domain lan
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using nameserver 127.0.0.1#5353
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: using 1 more local addresses
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: read /etc/hosts - 4 addresses
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq[3538]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Aug 18 09:56:44 2021 daemon.info dnsmasq-dhcp[3538]: read /etc/ethers - 0 addresses
Wed Aug 18 09:56:44 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Aug 18 09:56:44 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' is now down
Wed Aug 18 09:56:44 2021 daemon.debug dnsmasq[3538]: stopped listening on wan1(#170): 192.168.10.2 port 53
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' is disabled
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' has link connectivity
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' is enabled
Wed Aug 18 09:56:44 2021 kern.info kernel: [ 4765.431793] 8021q: adding VLAN 0 to HW filter on device wan1
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' is setting up now
Wed Aug 18 09:56:44 2021 daemon.debug dnsmasq[3538]: listening on wan1(#182): 192.168.10.2 port 53
Wed Aug 18 09:56:44 2021 daemon.notice netifd: Interface 'wan1' is now up
Wed Aug 18 09:56:44 2021 user.notice ModemManager: hotplug: add network interface wan1: event processed
Wed Aug 18 09:56:44 2021 user.notice ModemManager: hotplug: event reported: action=add, name=wan1, subsystem=net
Wed Aug 18 09:56:44 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Aug 18 09:56:44 2021 user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 206.189.3.13, no server API answer on 206.189.3.13)
Wed Aug 18 09:56:44 2021 user.notice OMR-VPS: Can't get vps token, try later
[... 223 lines cut ...]
Wed Aug 18 09:59:44 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 09:59:49 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 09:59:54 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 09:59:59 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:00 2021 cron.err crond[3735]: USER root pid 23340 cmd /etc/init.d/vnstat_backup backup
Wed Aug 18 10:00:04 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:09 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:14 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:19 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:24 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:29 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:34 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:39 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:44 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:49 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:00:55 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
Wed Aug 18 10:01:00 2021 daemon.err /usr/bin/ss-local[29246]: fast_open_connect: Network unreachable
````

## Specifications

  - OpenMPTCProuter version:openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-08-18:
It seems that nothing answer to ping.
Try to reboot OpenMPTCProuter, and check route table via SSH using ```ip r``` and ip config with ```ip a```

eothein, 2021-08-19:
Hey Yannick

See the output below. I was unsure on how to proceed as I don't know the internal working of MPTCPRouter. 

```
root@OpenMPTCProuter:~# ip r
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 1200
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.10.0/24 dev wan1 scope link metric 7
192.168.11.0/24 dev wan2 scope link metric 8
```

```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 100
    link/ether ba:8e:44:ed:8f:4f brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
3: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
4: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1464 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: bond0: <BROADCAST,MULTICAST,MASTER,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 22:1f:f7:39:7e:a5 brd ff:ff:ff:ff:ff:ff
10: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
132: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether fe:34:65:e0:f9:f4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.11.2/24 brd 192.168.11.255 scope global wan2
       valid_lft forever preferred_lft forever
182: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether da:1d:5b:78:ed:99 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.2/24 brd 192.168.10.255 scope global wan1
       valid_lft forever preferred_lft forever
191: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::6343:957b:433c:87c5/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
194: ifb4tun0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 26:9f:ac:78:9c:b8 brd ff:ff:ff:ff:ff:ff
```

Ysurac, 2021-08-19:
I don't see the VPS IP in the routes, so there is a problem somewhere. How did you configure the wizard ?

eothein, 2021-08-19:
Like normal: I surfed to 192.168.100.1, filled in the required input and saved the changes.

Ysurac, 2021-08-19:
Paste here the full config, from System->OpenMPTCProuter, "Show all settings" tab.

eothein, 2021-08-19:
It is a lot, but here it is. I hope I did not include too many secret details now?

```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].dnssec='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/zoom.us/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs7'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs4'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs4'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs4'
dscp.@classify[3].dest_port='65500'
dscp.@classify[3].comment='OMR API'
[... 1644 lines cut ...]
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'
```

Ysurac, 2021-08-19:
La configuration semble bonne.
Par contre il n'arrive visiblement pas à utiliser WAN1 et WAN2 pour aller sur internet. Le ping n'est pas bloqué sur ces wans ?
Que donne, via SSH sur OpenMPTCProuter, un ```ping -B -I wan1 1.1.1.1``` ?

eothein, 2021-08-19:
```
root@OpenMPTCProuter:~# ping -B -I wan1 1.1.1.1
PING 1.1.1.1 (1.1.1.1) from 192.168.10.2 wan1: 56(84) bytes of data.
```
Ne rien. It seems that it does not come through. 

Par contre:

```
root@OpenMPTCProuter:~# ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1) 56(84) bytes of data.
64 bytes from 192.168.10.1: icmp_seq=1 ttl=64 time=0.495 ms
64 bytes from 192.168.10.1: icmp_seq=2 ttl=64 time=0.445 ms
^C
--- 192.168.10.1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 0.445/0.470/0.495/0.025 ms
root@OpenMPTCProuter:~# ping 192.168.10.2
PING 192.168.10.2 (192.168.10.2) 56(84) bytes of data.
64 bytes from 192.168.10.2: icmp_seq=1 ttl=64 time=0.141 ms
64 bytes from 192.168.10.2: icmp_seq=2 ttl=64 time=0.114 ms
64 bytes from 192.168.10.2: icmp_seq=3 ttl=64 time=0.086 ms
^C
--- 192.168.10.2 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2082ms
rtt min/avg/max/mdev = 0.086/0.113/0.141/0.022 ms
root@OpenMPTCProuter:~# ping 192.168.11.2
PING 192.168.11.2 (192.168.11.2) 56(84) bytes of data.
64 bytes from 192.168.11.2: icmp_seq=1 ttl=64 time=0.144 ms
64 bytes from 192.168.11.2: icmp_seq=2 ttl=64 time=0.077 ms
^C
--- 192.168.11.2 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1050ms
rtt min/avg/max/mdev = 0.077/0.110/0.144/0.033 ms
root@OpenMPTCProuter:~# ping 192.168.11.1
PING 192.168.11.1 (192.168.11.1) 56(84) bytes of data.
64 bytes from 192.168.11.1: icmp_seq=1 ttl=64 time=0.488 ms
64 bytes from 192.168.11.1: icmp_seq=2 ttl=64 time=0.330 ms
64 bytes from 192.168.11.1: icmp_seq=3 ttl=64 time=0.397 ms
```
And if I log in to the Mikrotik controller I am able to ping 1.1.1.1 so it has a connection. 

```
[admin@MikroTik1] > ping 1.1.1.1
  SEQ HOST                                     SIZE TTL TIME  STATUS
    0 1.1.1.1                                    56  55 8ms
    1 1.1.1.1                                    56  55 7ms
    2 1.1.1.1                                    56  55 7ms
```
I am not able to ping 192.168.100.1 from the Mikrotik, but they are in other subnets so I guess that is normal.

Ysurac, 2021-08-19:
Did you reboot OpenMPTCProuter ?
Both Mikrotik can route the traffic ?

eothein, 2021-08-19:
- Both Mikrotik are able to route traffic
- OpenMPTCPRouter has been rebooted - even with a hard reboot. 

I'll try to change the MicroSD card, maybe something is going wrong there?

Ysurac, 2021-08-19:
If there is a problem with SD card it's indicated in status page.
Can you try to validate wizard again ?

eothein, 2021-08-19:
There is no problem indicated with the SD card. Tried to validate the wizard again (deleted the VPS and the wan1 and wan2 and re-added them). No luck.

eothein, 2021-08-19:
In the logs I find

```
Thu Aug 19 15:14:29 2021 user.notice OMR-VPS: Can't get VPSServer token, try later (can't ping server VPSServer on 206.189.3.13, no server API answer on 206.189.3.13)
Thu Aug 19 15:14:31 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because check error and ping from 192.168.11.100 error (4.2.2.1,8.8.8.8,80.67.169.12)
Thu Aug 19 15:14:31 2021 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because check error and ping from 192.168.10.100 error (4.2.2.1,8.8.8.8,80.67.169.12)
Thu Aug 19 15:14:31 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Thu Aug 19 15:14:31 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev wan1
```

So it seems the default route gets deleted for some reason? Moreover, when I rebooted everything I saw I the first routing table I asked for

```
default 192.168.10.1 0.0.0.0 UG 3 0 0 wan1
```
But the second try I don't see the default route anymore?

Kind regards
Jens Buysse

Ysurac, 2021-08-19:
Default route are removed because connection can't be used: ping from wan interface doesn't work.

eothein, 2021-08-19:
After re-installing OpenMPTCPRouter, disabling the MiktroTeks and setting everything back up again, I finally got the setup working. No idea where the mistake was...

eothein, 2021-08-20:
I finally found the issues: it had nothing to do with OpenMPTCPRouter itself, but the Mikrotek Configuration: it seems the NAT was net enabled, hence pinging worked, but there was no mapping between the subnets.

In any case, thanks for the help!

closed 2021-08-20 by eothein