Ysurac/openmptcprouter #2035: Wifi bridging on Raspberry 4 with macvlan

highfreq, 2021-08-21:
## Expected Behavior
Raspberry 4 with 2 macvlan routers on lan port and lan port connected to 2 local pc trough a switch. Would like to have wifi access to local lan to reach the 2 pc (i guess wifi bridge to local lan)

## Current Behavior
If i setup wifi access point and bridge it to lan than the macvlan interfaces say "cn't claim device", i guess it is because lan0 is now bridged.

## Specifications

 Latest OMR on Raspberry 4

highfreq, 2021-08-22:
No chance to have a little help?

highfreq, 2021-08-24:
Guess not :)

Ysurac, 2021-08-24:
Please use the full template and set correctly the release you use in specifications.
With macvlan I don't think bridge can work. But OpenMPTCProuter is a router, they don't need to be on same LAN to reach each others, you only need to set firewall correctly.

highfreq, 2021-08-24:
Release is latest, installed 3 days ago.
So you mean i just activate wifi network and put it in lan zone of firewall?

Ysurac, 2021-08-24:
As indicated in template, latest is not a release. And this doesn't help when someone check the issue 3 months later...

highfreq, 2021-08-24:
Ok, sorry about it, edited first post.

Ysurac, 2021-08-24:
This should work: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

highfreq, 2021-08-24:
Thank you, i followed it step by step and i can now connect to my wireless network, wich was added in the lan zone of the router. 
I can now ping router lan address but none of the pc attached to the lan itself.

Ysurac, 2021-08-24:
You can try to create a new firewall zone for wifi interface, set it with same parameters as LAN zone and allow forward from this zone to LAN.

highfreq, 2021-08-24:
done firewall zone same as lan but still can't ping lan pc. Can only ping router lan address.

Ysurac, 2021-08-24:
Forward is allowed in firewall zone configuration?

highfreq, 2021-08-24:
yes, all allowed

highfreq, 2021-08-24:
wifi ---> lan      allowed allowed allowed

highfreq, 2021-08-25:
Made a clean install and followed the guide and now everything works. 
Thank you very much.

Ysurac, 2021-08-25:
Thanks for the feedback.

closed 2021-08-25 by Ysurac