Ysurac/openmptcprouter #2038: Burn an image on new Raspberry 4 and connect to a new server.

highfreq, 2021-08-25:
Load an SD image on a new installation and connect it to a different server changing IP and keys, would it work? I have some port forwarding configured, would it keep it?
I would like to do it so that i can replicate my perfectly now functioning OMR with all the configurations i need.

## Current Behavior
Did not try yet

  - OpenMPTCProuter version: OMR v0.58.5 on Raspberry 4
  - OpenMPTCProuter VPS version: OMR v0.58.5 on Ubuntu 20
  - OpenMPTCProuter VPS provider: Aruba
  - OpenMPTCProuter platform: Rpi4

Ysurac, 2021-08-25:
Yes this should work.

highfreq, 2021-08-25:
OMR will forward port forwarding configuration to new server trough api?

Ysurac, 2021-08-25:
Yes. It's it's not done after wizard validation, this should be done a few minutes later.

highfreq, 2021-08-25:
Ok great, i'll try it.

Thank you very much.

closed 2021-11-28 by github-actions[bot]