Ysurac/openmptcprouter #2045: No aggregation in download

fredb80, 2021-08-28:
## Expected Behavior
Aggregation in upload and downlond

## Current Behavior
Following a 'readonly' error on a rapsberry pi 3 sd card. I have re-installed with the latest version and since then I no longer have an aggregation in download only in upload
## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d 
  - OpenMPTCProuter VPS version: v0.1026 5.4.100
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-08-31:
How do you test download ? Can I get the result of "omr-test-speed" via SSH on the router ?

wlatic, 2021-09-02:
How many connections do you have? Also any disabled or backup connections?

fredb80, 2021-09-02:
Thank for your response:

omr-test speed on the router

Select best test server...
host: speedtest.frankfurt.linode.com - ping: 33
host: speedtest.tokyo2.linode.com - ping: 267
host: speedtest.singapore.linode.com - ping: 259
host: speedtest.newark.linode.com - ping: 117
host: speedtest.atlanta.linode.com - ping: 121
host: speedtest.dallas.linode.com - ping: 149
host: speedtest.fremont.linode.com - ping: 163
host: speedtest.tele2.net - ping: 34
host: proof.ovh.net - ping: 23
host: speed.hetzner.de - ping: 31
host: ipv4.bouygues.testdebit.info - ping: 25
host: par.download.datapacket.com - ping: 25
host: nyc.download.datapacket.com - ping: 104
host: ams.download.datapacket.com - ping: 30
host: fra.download.datapacket.com - ping: 33
host: lon.download.datapacket.com - ping: 37
host: mad.download.datapacket.com - ping: 44
host: prg.download.datapacket.com - ping: 35
host: sto.download.datapacket.com - ping: 55
host: vie.download.datapacket.com - ping: 48
host: war.download.datapacket.com - ping: 48
host: atl.download.datapacket.com - ping: 118
host: chi.download.datapacket.com - ping: 114
host: lax.download.datapacket.com - ping: 162
host: mia.download.datapacket.com - ping: 128
host: nyc.download.datapacket.com - ping: 92
Best server is http://proof.ovh.net/files/10Gb.dat, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0   2472      0 --:--:-- --:--:-- --:--:--  2507

And I've test speed by N-perf

fredb80, 2021-09-02:
[image]

JulesCatherine, 2021-11-06:
Hi you haven't found a solution ?
Because it looks like i have the same problem

closed 2022-02-10 by github-actions[bot]