Ysurac/openmptcprouter #2047: Redirection

Pokpok-sbo, 2021-08-31:
Hello everyone,

Here is my problem:

I have an Ubuntu server (192.168.1.1) which manages my various internet connections, DHCP and my local DNS.
Until now I had a second server which only managed a website (192.168.1.10)
To access it from outside with the URL test1.pok.com, I simply redirected port 80 in 192.168.1.1 to 192.168.1.10

Everything was going very well.

BUT now I have a third machine (192.168.1.20) which runs a second website
I created a new URL test2.pok.com. In 192.168.1.1, I created a DNS that points this URL to 192.168.1.20.
So when I am in my local network it works fine.

However, from outside when I try to go to my test2.pok.com site I land on my first site (test1.pok.com).

My question is simple, what should I configure in my 192.168.1.1 server so that from the outside my two URLs point to the right machine in my network?

Thank you in advance for your answers

Ysurac, 2021-08-31:
Port can only be redirected to one server. You can use a nginx as proxy to redirect based on domain name on 192.168.1.1

Pokpok-sbo, 2021-08-31:
Thanks for the tip. 
Looking in this direction, I found this tutorial that matches my request:
[](url)https://www.it-connect.fr/configurer-nginx-en-tant-que-reverse-proxy/

Hope it will be useful to others

closed 2021-08-31 by Pokpok-sbo