Ysurac/openmptcprouter #2048: After being idle on facebook: NET::ERR_CERT_COMMON_NAME_INVALID

dennydeutscher, 2021-08-31:
## Expected Behavior

Facebook shall not close the connection nor spit out SSL Errors

## Current Behavior
After being idle on facebook: The connection gets closed after 10 minutes. Trying to 
reconnect (re-fresh) fails with an NET::ERR_CERT_COMMON_NAME_INVALID Error

## Possible Solution

Restarting the PC solves the problem for the next few minutes.

## Steps to Reproduce the Problem

  1. On Windows 10 with Edge Browser open facebook and log-in
  2. Leave the Browser window in the background
  3. after a while the message "you're offline" appears. Refreshing the page results in the error mentioned before. 
  4. over my default (network) internet connection everything works fine with the same pc (using openmptcprouter only 
  for evaluating purposes right now). 

## Context (Environment)

Leaving Facebook in background to receive messages. 

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS provider: Hetzner.de
  - OpenMPTCProuter platform: WRT3200ACM
  - Country: Germany

Ysurac, 2021-08-31:
All is green in status page ? Output ip is VPS public IP ?

dennydeutscher, 2021-09-01:
> All is green in status page ? Output ip is VPS public IP ?

Yes, except one of the LTE connection is switching to "no ping response (orange)" sometimes -due to low cellular network condition at the location I'm testing.

Ysurac, 2021-09-01:
Can you check "State->System log" for error with shadowsocks (if you use the default proxy) ?

dennydeutscher, 2021-09-01:
No, there is no error in dependency with shadowsocks... I've the the issue with v2ray too btw.

dennydeutscher, 2021-09-02:
Me again: I think I solved the problem with facebook ssl errors. 

I had additional DNS servers configured (for each wan/wwan connection the depending DNS). 
I removed the 127.0.0.1#5353 default entry end entered the new servers (just like you describe in the [wiki](https://github.com/Ysurac/openmptcprouter/wiki/Custom-DNS))

After reverting to default settings everything seems to be good so far. Was it my fault in some way?