Ysurac/openmptcprouter #2049: Connections in Order - Need to have Active, then backup or having issues

wlatic, 2021-09-01:
## Expected Behavior

You are able to add interfaces in any order and you can enable / disable / set them as backup at any point in time.

## Current Behavior

I had Connection 1 Enabled, Connection 2 Master, Connection 3 backup, Connection 4 enabled, Connection 5 disabled.

Was getting strange errors such as:
OMR-VPS: Can't get VPS token
Shadown Socks / OMR constantly rebooting

## Possible Solution

I re-arranged my connections so active were the first 3, backup was 4 and disabled was 5

## Steps to Reproduce the Problem

  1. Add 2 active connections
  2. Add 1 backup connection
  3. Add 1 active connection
  4. Add 1 disabled connection
  
  Check logread for errors

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version:  0.1026
  - OpenMPTCProuter VPS provider: RackNerd
  - OpenMPTCProuter platform: x86_64
  - Country: USA

Ysurac, 2021-09-03:
One connection must be set as master, else you can have some strange result.
You can check log in State->System log to know what is the issue.

wlatic, 2021-09-04:
Couldn't get it working as just a backup and on checking if it'd work as "enabled" it did so without a problem.

When then swapping it back over to backup it worked without an issue.

Will test again after a reboot.

wlatic, 2021-09-04:
Im wondering if this is because the backup connection has 12 ping and rest are much higher (LTE).

I've done a whole reinstall and recofigure, keeping it pretty simple and the same results happened, lots of issues connecting and restarting if the backup connection is present. As soon as you set it to Enabled and reboot it comes back without any issues.

I was seeing if I could set a limit for the DSL connection over the tunnel and give it 50kb/s down and 15kb/s up and then it shouldn't affect the transaction much (the connection is only 9000kbps down 900kbps up) and will stop these issues, but it doesn't look like that is possible.

wlatic, 2021-09-08:
Main problem with this approach appears to be that because the connection has a very low ping if I leave OMR-bypass to figure it out itself it will normally use the DSL connection for everything.

@Ysurac are you seeing any errors when using backup connections elsewhere?

closed 2021-12-13 by github-actions[bot]