Ysurac/openmptcprouter #2050: Configuration using CLI

eothein, 2021-09-02:
I wish to use OpenMPTCPRouter dynamically, is there a CLI with corresponding commands which can be used on the router? Is there any documentation for this? 

This way I can easily play around with different setups, without loading the web interface.

highfreq, 2021-09-04:
I think you can do pretty much everything with luci commands from cli.

eothein, 2021-09-06:
I'm not sure what luci commands are, is there some documentation somewhere (with examples?)

Ysurac, 2021-09-06:
```ubus``` and ```uci``` commands are available on the router. There is some documentation on OpenWrt website.
For now all is not yet available via ubus, but all configuration can be modified via uci.

closed 2021-12-10 by github-actions[bot]