Ysurac/openmptcprouter #2051: Question: Why android tethering triggers adb request?

storm1er, 2021-09-02:
Hi,

This is a simple question: when plugging in my android to add a connection, it triggers an adb request.
Why openmtcp would request an adb access?

I see it's not needed as I never accepted and the connection works, just wondering.

eothein, 2021-09-02:
I don't think this has something to do with OpenMPTCPRouter: if your Android phone is in Developer modus, plugging in an USB cable will always trigger an adb request.

storm1er, 2021-09-02:
You can easily test it with a windows laptop.

It will be triggered only after an adb server start (or any sub command that need a server start)

storm1er, 2021-09-02:
Unless there's an adb server permanently running which feels kinda weird :/

storm1er, 2021-09-02:
and will it add more features or data if I accept the request using my phone?

closed 2021-09-02 by storm1er