Ysurac/openmptcprouter #2052: Connection loss when reconfiguring

eothein, 2021-09-02:
Dear all

I'm benchmarking OpenMPTCPRouter: my setup is one Router, three MikroTekRouters which serve as WAN access, but where I perform some traffic shaping (max. down/up Bitrate is set to 10 Mbps). I have one NUC client who serves as the client for the iperf3 tests. 

My test case is the following: in the wizard, I enable only the master MikroTik Router.  I start the iperf3 test for a long time and after I minute I reconfigure the MPTCP router and enable the second MikroTik Router.  I notice that I get the following error when reconfiguring OpenMPTCPRouter:

``iperf3: error - control socket has closed unexpectedly ``

Is this to be expected? I imagined that the MPTCP connection would still remain working, but now would aggregate its traffic over both links? I want to achieve the following graph (but then continuous - without interruption) -notice the error in the y-label (should be x 10).  

[image]

Ysurac, 2021-09-02:
What you mean by "reconfigure the MPTCP router" ?

eothein, 2021-09-02:
I have a follow -up question: instead of reconfiguring, I'm now pulling out cables to mimic outages in connections (e.g. one of the Routers is out of LTE range). I would suspect that Bitrate goes down, and when I plug back in the cable it should go up, however I get the following output:

````
Connecting to host iperf.par2.as49434.net, port 9201
[  5] local 192.168.100.226 port 48544 connected to 193.177.162.41 port 9201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  6.50 MBytes  54.5 Mbits/sec    0    250 KBytes
[  5]   1.00-2.00   sec  5.90 MBytes  49.5 Mbits/sec    0    389 KBytes
[  5]   2.00-3.00   sec  2.98 MBytes  25.0 Mbits/sec    0    389 KBytes
[  5]   3.00-4.00   sec  3.54 MBytes  29.7 Mbits/sec    0    407 KBytes
[  5]   4.00-5.00   sec  2.98 MBytes  25.0 Mbits/sec    0    407 KBytes
[  5]   5.00-6.00   sec  3.36 MBytes  28.1 Mbits/sec    0    407 KBytes
[  5]   6.00-7.00   sec  3.91 MBytes  32.8 Mbits/sec    0    407 KBytes
[  5]   7.00-8.00   sec  3.04 MBytes  25.5 Mbits/sec    0    407 KBytes
[  5]   8.00-9.00   sec  4.04 MBytes  33.9 Mbits/sec    0    397 KBytes
[  5]   9.00-10.00  sec  3.23 MBytes  27.1 Mbits/sec    0    438 KBytes
[  5]  10.00-11.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  11.00-12.00  sec  2.17 MBytes  18.3 Mbits/sec    0    438 KBytes
[  5]  12.00-13.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  13.00-14.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  14.00-15.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  15.00-16.00  sec  2.30 MBytes  19.3 Mbits/sec    0    438 KBytes
[  5]  16.00-17.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  17.00-18.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  18.00-19.00  sec  2.17 MBytes  18.3 Mbits/sec    0    438 KBytes
[  5]  19.00-20.00  sec  2.30 MBytes  19.3 Mbits/sec    0    438 KBytes
[  5]  20.00-21.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  21.00-22.00  sec  2.17 MBytes  18.2 Mbits/sec    0    438 KBytes
[  5]  22.00-23.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  23.00-24.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  24.00-25.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  25.00-26.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  26.00-27.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  27.00-28.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  28.00-29.00  sec  2.17 MBytes  18.2 Mbits/sec    0    438 KBytes
[  5]  29.00-30.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  30.00-31.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  31.00-32.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  32.00-33.00  sec  2.30 MBytes  19.3 Mbits/sec    0    438 KBytes
[  5]  33.00-34.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  34.00-35.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  35.00-36.00  sec  2.17 MBytes  18.3 Mbits/sec    0    438 KBytes
[  5]  36.00-37.00  sec  2.24 MBytes  18.8 Mbits/sec    0    438 KBytes
[  5]  37.00-38.00  sec  29.0 MBytes   244 Mbits/sec    0    646 KBytes
[  5]  38.00-39.00  sec  2.50 MBytes  21.0 Mbits/sec    0    785 KBytes
[  5]  39.00-40.00  sec  0.00 Bytes  0.00 bits/sec    0    785 KBytes
[  5]  40.00-41.00  sec  11.2 MBytes  94.4 Mbits/sec    0    933 KBytes
[  5]  41.00-42.00  sec  0.00 Bytes  0.00 bits/sec    0    933 KBytes
[  5]  42.00-43.00  sec  0.00 Bytes  0.00 bits/sec    0    933 KBytes
[  5]  43.00-44.00  sec  0.00 Bytes  0.00 bits/sec    0    933 KBytes
[  5]  44.00-45.00  sec  3.75 MBytes  31.5 Mbits/sec    0    707 KBytes
[  5]  45.00-46.00  sec  6.25 MBytes  52.4 Mbits/sec    0   1.01 MBytes
[  5]  46.00-47.00  sec  0.00 Bytes  0.00 bits/sec    0   1.01 MBytes
[  5]  47.00-48.00  sec  0.00 Bytes  0.00 bits/sec    0   1.01 MBytes
[  5]  48.00-49.00  sec  0.00 Bytes  0.00 bits/sec    0   1.01 MBytes
[  5]  49.00-50.00  sec  10.0 MBytes  84.0 Mbits/sec    0   1.06 MBytes
[  5]  50.00-51.00  sec  0.00 Bytes  0.00 bits/sec    0   1.06 MBytes
[  5]  51.00-52.00  sec  0.00 Bytes  0.00 bits/sec    0   1.06 MBytes
[  5]  52.00-53.00  sec  0.00 Bytes  0.00 bits/sec    0   1.06 MBytes
[  5]  53.00-54.00  sec  0.00 Bytes  0.00 bits/sec    0   1.06 MBytes
[  5]  54.00-55.00  sec  11.2 MBytes  94.4 Mbits/sec    0   1.11 MBytes
[  5]  55.00-56.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  56.00-57.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  57.00-58.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  58.00-59.00  sec  5.00 MBytes  42.0 Mbits/sec    0    897 KBytes
[  5]  59.00-60.00  sec  5.00 MBytes  41.9 Mbits/sec    0   1.11 MBytes
[  5]  60.00-61.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  61.00-62.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  62.00-63.00  sec  0.00 Bytes  0.00 bits/sec    0   1.11 MBytes
[  5]  63.00-64.00  sec  11.2 MBytes  94.5 Mbits/sec    0   1.11 MBytes
````

Ysurac, 2021-09-02:
Wizard reset all connections.
Check without unplugging master connection and by downloading a file via curl or wget.
iperf sometimes give strange results.

closed 2021-12-06 by github-actions[bot]