Ysurac/openmptcprouter #2053: Best configuration to have 2 LTE modems always up

highfreq, 2021-09-02:
I would like to know which is the best configuration in order to have the connection never down of 2 lte modems. 
At the moment i have it configured as per default, but the connection stays up if the slave falls, but when the master falls connection goes down.
Should i configure the two modems as redundant and have a master and a backup?

I am a bit lost at the moment, help would be greatly appreciated.

My main goal is to have aggregation but most important is to never have the line goes down.

wlatic, 2021-09-02:
On your advanced settings what is your "Master interface selection" setting?

highfreq, 2021-09-03:
Is set to balancing

highfreq, 2021-09-03:
Any help would be greatly appreciated, i would really love to read documentation but there isn't much around. Read all the avail docs on the wiki but 99% of the functions are not documented.

Ysurac, 2021-09-03:
Master interface selection definite how routes are defined. Balancing means routes are in load balancing mode.
Check State->System log to know why you have a disconnection

highfreq, 2021-09-03:
I don't have a disconnection, but if i take down master wan interface the link goes down while if i take down slave than the link stays up. Is there a way to make it not go down if i take down master wan?

closed 2021-12-07 by github-actions[bot]