Ysurac/openmptcprouter #2054: New forwarded ports configuration not pushed to primary VPS

sebvallee, 2021-09-02:
## Expected Behavior

Having new forwarded ports configuration pushed to all VPS in OMR configuration.

## Current Behavior

The new forwarded ports configuration is only pushed to the first VPS in the list, despite the master VPS is the second one.
Re-validating the wizard, with no modification on it, solve the problem, until next addition to the forwarded ports configuration.

## Possible Solution

Having new forwarded ports configuration pushed to all VPS in OMR configuration at creation, no matter which VPS is set as master.

## Steps to Reproduce the Problem

- Add 2 VPS on your OMR configuration
- Set second VPS on the list as master VPS
- Add a forwarded port on the OMR config
- Check on both VPS, the rule is only on the first VPS, which is not the master one

## Context (Environment)

Not sure what to fill here.

## Specifications

- OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
- OpenMPTCProuter VPS version: 0.1026
- OpenMPTCProuter VPS provider: Milkywan / Sapinet (not relevant to this issue)
- OpenMPTCProuter platform: x86_64
- Country: FR

Ysurac, 2021-09-02:
What is the result, via SSH on the router, of ```/etc/init.d/openmptcprouter-vps set_vps_firewall``` and do you have error in log (```logread```) ?

sebvallee, 2021-09-02:
No output on `/etc/init.d/openmptcprouter-vps set_vps_firewall` (exit code 0) and no error seen on `logread`

sebvallee, 2021-09-02:
I forgot to mention but it's a fresh install (both vps and router)

closed 2021-12-07 by github-actions[bot]