Ysurac/openmptcprouter #2055: SIP Phone Audio Problem

fzonno, 2021-09-03:
## Expected Behavior
Remote SIP phone
We have a SIP PBX (3CX) and some remote phone (STUN mode).

## Current Behavior
The remote phones register but there is no audio communication 
The phones use UDP/TCP 5060 for register and UDP ports 9000-10999 for audio
We have a pfSense firewall behind the openmtcprouter with a forward rule any/any.
The PBX ports test shows that there is a SIP ALG service active. It can be the problem. Any hints for deactivate it?
Any other service / port works perfectly (OpenVPN / HTTPS / SMTP / FTP etc.)

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026.5.4.100
  - OpenMPTCProuter VPS provider: ServerPlan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-09-03:
You can try to disable "Automatic helper assignment" on VPN and WAN zone in firewall configuration.

fzonno, 2021-09-07:
I disabled the Automatic helper assignment but the problem persists.
Any other clue?

Ysurac, 2021-09-07:
No. To my knowledge there is no ALG service on OpenMPTCProuter or OpenWrt.

fzonno, 2021-09-07:
I tryed without the pfsense (pbx directly connected to openmptcp but same result.
This is the result of the firewall check test
[image]

Ysurac, 2021-09-07:
Maybe the problem is on VPS side.
Can you try to remove sip related modules: ```modprobe -r nf_nat_sip``` and ```modprobe -r nf_conntrack_sip```

closed 2021-12-11 by github-actions[bot]