Ysurac/openmptcprouter #2057: OMR Sur mini routeur GL-iNet GL-AR750

shinjii, 2021-09-04:
Bonjour a tous,
Est il possible d'installer OMR sur un  GL-iNet GL-AR750?

je suppose qu'il faut créer (si c 'est possible) une image pour celui ci.
je n'ai pas bien compris la section unsupported platform, si une personne a un peux de temps pour m'expliquer comment faire ca serais top (en Français de préférence, je suis pas très bon avec l'anglais '-_-)

Merci d'avance pour vos retour :)

Adorfer, 2021-09-04:
I would not expect more than 15-20MBit/s at best as total bandwidth with that device, with USB-LTE dongles (cpu intensive io load) even less. 
I other words: This may be sufficient for your case, but will probably not motivate a lot of people to put effortds into it.

shinjii, 2021-09-07:
Ok , je note , merci ;)

closed 2021-12-11 by github-actions[bot]