Ysurac/openmptcprouter #2060: Robust 2 LTE modems connection

highfreq, 2021-09-05:
Is there a way with 2 lte modems to have the connection stay up if the master modem loses connection?

If slave connection is lost main connection stays up but if master goes down i lose it all. What should i do to have them stay up if master goes down?

donbecker, 2021-09-18:
I have this exact same issue. I have 3 LTE modems connected. 
One is set to Multipath TCP: Master
The other two are set to Multipath TCP: Enabled

Is there any way to have the Master role switch to one of the Enabled if the Master goes down?

Ysurac, 2021-09-18:
@donbecker No need to switch master, if master go down others connection should take the relay. Connection can be down when master is down for a short time because of some packet lost. MPTCP retry in Network->MPTCP can help.

highfreq, 2021-09-18:
If master goes down it takes 20-30 seconds for my router to start using the other connection. Is this a normal switch time?

Ysurac, 2021-09-18:
Check log.

highfreq, 2021-09-18:
This is the 2 lines of log after switching off wan1 (master). Looks like wan2 goes offline too because it can't ping, but in reality it does ping.

Sat Sep 18 11:54:11 2021 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because check error and ping from 192.168.10.2 error (1.1.1.1,4.2.2.1,8.8.8.8)
Sat Sep 18 11:54:11 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev wan1

donbecker, 2021-09-20:
The few times that Master has gone offline, I lose all connectivity and it never seems to switch to using the remaining connections, I'll work on testing this further this week.

gre2gor, 2021-09-30:
Anyone had succes on this?

Ysurac, 2021-09-30:
When there is a problem, all infos are in system logs. So if master go down, you should see if VPN and proxy need to reconnect or not.

highfreq, 2021-09-30:
> Anyone had succes on this?

I did not, from system log i can't see where my problem is. All i know is that if master goes down than all link goes down, partially defeating the reason i am using OMR in the first place.

Ysurac, 2021-09-30:
Do you enabled SQM and set download/upload speed ?

highfreq, 2021-09-30:
I didn't, does it have anything to do with link robustness?

Ysurac, 2021-09-30:
SQM can avoid link saturation, so if ping doesn't work because all bandwidth is used this can help.

highfreq, 2021-09-30:
It is not saturated at all, link just goes down when master link falls

Ysurac, 2021-09-30:
You see all links down in system log with ping errors ?

highfreq, 2021-09-30:
Sat Sep 18 11:54:11 2021 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because check error and ping from 192.168.10.2 error (1.1.1.1,4.2.2.1,8.8.8.8)
Sat Sep 18 11:54:11 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev wan1

Ysurac, 2021-09-30:
This doesn't really help to see an issue. wan1 is detected as down, I think you have more links than that.

highfreq, 2021-09-30:
I have wan1 and wan2 on 2 lte modems.

Ysurac, 2021-09-30:
So you said all links go down, you should have more logs with also wan2 down and VPN/Proxy down too.

highfreq, 2021-09-30:
I said, when master goes down the slave link is unable to carry on with comunication, it tkes 20-30 seconds for link slave to pick up the link and start sending data. 
To make it short, if master goes down there is a link blackout of around 20-30 seconds, than comunication is restblished.

Ysurac, 2021-09-30:
Again, check in log to find what go down: VPN or proxy. And you can try another VPN like glorytun UDP that may have no problem with master down.

highfreq, 2021-09-30:
Ok, will try thank you

closed 2022-01-03 by github-actions[bot]