Ysurac/openmptcprouter #2062: OMR VPS: help to use it for hosting other websites

nicolasfruit, 2021-09-06:
Dear all,

First I would like to thanks all contributors and specifically @Ysurac for his amazing work on OMR !

I have setup a fully functional OMR config (raspberry at home + OVH VPS) and agregate ADSL + 2x 4G LTE (in a rural zone I increase by x10 my ADSL speed !)

I would like to use my VPS for other web hosting. May I have some guidance to allow indbound trafic (typically Web port 443 + 80) on the OMR VPS. It seems that with the VPS install script of OMR everything is blocked now ...

Regards
Nicolas.

Ysurac, 2021-09-06:
On the VPS shorewall is used. To open a port you need to modify ```/etc/shorewall/rules``` and restart or reload shorewall (```systemctl restart shorewall```). You can use SSH rule as an example.

nicolasfruit, 2021-09-06:
Thanks @Ysurac for your quick answer. **It works** !
I added the following rules and restart :
```
#
ACCEPT          net             $FW             tcp     80
ACCEPT          net             $FW             tcp     443
```

Thx,
Nicolas.

closed 2021-09-06 by nicolasfruit