Ysurac/openmptcprouter #2068: Exclude WAN Connection from OMR Bypass

wlatic, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
Using OMR-bypass is great, however there are sometimes in which I'd rather not choose which connection is being used, but exclude a connection from being used.

This would mean I could exclude my DSL connection from being used by OMR-Bypass, unless it was physically selected to be used (It would stop my TVs and streaming devices using the DSL connection)

**Describe the solution you'd like**
Option to Exclude Interfaces on OMR-Bypass interface selection screen

**Describe alternatives you've considered**
Tried to use the DSL connection as a backup, however this is having issues

Ysurac, 2021-09-09:
I don't see real use cases for that.

wlatic, 2021-09-09:
When you have people working from home they may want to anchor themselves to a particular connection, and also have TVs / Steaming services which they are using OMR Bypass for.

In an ideal world I want OMR bypass to be able to select which connection is used for the TVs (that doesn't matter), but ensuring the Tv's dont use the DSL line used for work is huge (especially when on a conference call with a 6mbps line).

wlatic, 2021-09-09:
Thought of another use case.

You have a network which uses gaming consoles and dont want them to be affected by Streaming from TVs etc.

A lot of people using openMPTCP are people who have very small DSL connections and looking to use LTE to get something usable.

closed 2021-12-14 by github-actions[bot]