Ysurac/openmptcprouter #2069: Support: Use adb and scrcpy

storm1er, 2021-09-09:
Hi,

I used to use scrcpy to see my phone screen as it was connected to my laptop as a USB modem.

Now that I use openmtcp as router with multiple 4G modem and sometimes my phone to add more bandwidth, I would like to see my phone screen on my laptop again while my phone is connected on the router.

To make scrcpy work, I need a working adb connection from my laptop to my phone.

So, basically, the flow looks like this:
```
phone (adb) ---usb--- openmtcp (adb forward + port forward?) ---lan--- laptop(adb connect remote + scrcpy)
```

The current problem is `adb forward dev:xxxxxx tcp:5555` on openmtcp always return "error: cannot bind to socket", whatever the port I try to use.

storm1er, 2021-09-09:
more logs from adb:

<details>

```
 ADB_TRACE=1 adb forward dev:xxxxxxxx tcp:5555
adb.c::main():Handling commandline()
adb_client.c::_adb_connect():_adb_connect: host:version
transport.c::writex():writex: fd=3 len=4: 30303063 000c
transport.c::writex():writex: fd=3 len=12: 686f73743a76657273696f6e host:version
transport.c::readx():readx: fd=3 wanted=4
transport.c::readx():readx: fd=3 wanted=4 got=4
4f4b4159 OKAY
adb_client.c::_adb_connect():_adb_connect: return fd 3
adb_client.c::adb_connect():adb_connect: service host:forward:dev:xxxxxxxx;tcp:5555
transport.c::readx():readx: fd=3 wanted=4
transport.c::readx():readx: fd=3 wanted=4 got=4
30303034 0004
transport.c::readx():readx: fd=3 wanted=4
transport.c::readx():readx: fd=3 wanted=4 got=4
30303230 0020
adb_client.c::_adb_connect():_adb_connect: host:forward:dev:xxxxxxxx;tcp:5555
transport.c::writex():writex: fd=3 len=4: 30303232 0022
transport.c::writex():writex: fd=3 len=34: 686f73743a666f72776172643a646576 host:forward:dev
transport.c::readx():readx: fd=3 wanted=4
transport.c::readx():readx: fd=3 wanted=4 got=4
4641494c FAIL
transport.c::readx():readx: fd=3 wanted=4
transport.c::readx():readx: fd=3 wanted=4 got=4
30303135 0015
transport.c::readx():readx: fd=3 wanted=21
transport.c::readx():readx: fd=3 wanted=21 got=21
63616e6e6f742062696e6420746f2073 cannot bind to s
adb_client.c::adb_connect():_adb_connect error: cannot bind to socket
adb_client.c::adb_connect():adb_connect: return fd -1
error: cannot bind to socket
error: cannot bind to socket
```

</details>

Ysurac, 2021-09-09:
adb is running in server mode by default, this may be the problem.

closed 2021-09-10 by storm1er