Ysurac/openmptcprouter #2072: Debugging flaky network connection

herrbuerger, 2021-09-10:
Hello,

I've been running OpenMPTCProuter for a few months now and aside from a few hickups never had any issues - until recently.

I'm running the following setup:

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter VPS provider: AWS EC2
  - OpenMPTCProuter platform: Raspberry PI 4B (64-bits)
  - WAN 1 - cable connection 
  - WAN 2 - 4G Vodafone connection
  - all settings default

Recently I have been getting the error message "multipath seems to be blocked on the connection" for the 4G connection. This has never happened before. I ran the MPTCP check and it really turns out that the connection is not supporting multipath anymore. **Has the provider really made a change to his system or could this be an issue with OpenMPTCProuter?**

Since that issue has shown up I have activated "MPTCP over VPN" for the 4G connection. This has solved the issue and aggregation is working fine again. However, I'm suddenly seeing lots of network interruptions (2-10 times per hour) - didn't happen before with normal MPTCP.
The signal strength is at 100% in the LTE router and the connection is generally very stable.

To dig further I ran a PING check from a different machine directly connected to the LTE modem. There were no issues with the ping nor the latency:

```
1061 packets transmitted, 1061 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 24.932/28.826/365.783/17.140 ms
```

However OpenMPTCProuter was again reporting multiple downtimes. The logs for that particular timeframe can be found here:
https://gist.github.com/herrbuerger/0dd8feea2cf00601ccf625a37328cae9

**Is there anything I can change in the configuration to make the connection more stable?**

And last but not least, I'd love to try out Glorytun UDP or MLVPN. However the wiki is quite vague on how to activate this. I've been trying lots of different combinations but couldn't get aggregation working again. 
**Can someone explain the necessary changes in a little more detail?**

Thanks for reading my wall of text and if I can provide any further information, please let me know.

--Chris

Ysurac, 2021-09-10:
Please use the template.

wlatic, 2021-09-11:
@herrbuerger when you check logread are you seeing any of the services restart or disconnects from the VPS when these issues happen?

herrbuerger, 2021-09-11:
@wlatic [the log I posted above](https://gist.github.com/herrbuerger/0dd8feea2cf00601ccf625a37328cae9) is exemplary for what happens during those disconnects. When checking logread on the console I don't see any additional information. As far as I can tell there is no disconnect from the VPS, only messages that the second connection is down. Usually the first log indicating a disconnect is `No answer from wgwan2 (wgwan2), restart interface`.

herrbuerger, 2021-09-24:
For anyone experiencing a similar issue: I had the problem that the LTE connection was constantly switching between 4G and 4G+. Once I set the LTE connection to use a fixed band the disconnects were reduced.

However, the connection was still too unstable. I have since switched to a commercial product which seems to have solved my problems.

closed 2021-09-24 by herrbuerger