Ysurac/openmptcprouter #2073: Recommended device for 2x 300Mbs ISP

bambuca, 2021-09-11:
Can you please recommend a suitable device? I have two providers, each with 300/300 Mbps. I currently have RPI4, but there I get speeds "only" around 500 Mbps.

gsxryan, 2021-09-12:
Do both ISP regularly test 300/300?  This may be ISP not giving full stated speed.

I was able to test pi4 no encryption at 600/690.  running macvlan single eth port.

If you want device running encryption running AES GCM, i'd recommend a processor that supports AES-NI.  There may be some SBC x86 boards available, but i have no testing done with these yet.

bambuca, 2021-09-13:
Yes, when I measure them separately, one has about 280 Mbps and the other 310 Mbps. I'm gonna look for some AES-NI capable device. Thanks.

closed 2021-09-13 by bambuca

ioogithub, 2022-09-14:
@bambuca what device did you end up getting? How has your experience been?