Ysurac/openmptcprouter #2076: Building images for ipq40xx fails because of BBR2 missing

debiantippse, 2021-09-14:
Hey i'm currently trying to Build images for AVM Devices(Fritzbox 7530)

When i try to build an image it actually gaves me after a while this error here:
[image]

Also at the beginning it starts with:

domenik@debian:~/openmptcprouter/ipq40xx/source$ scripts/feeds install -a -d y -f -p openmptcprouter
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist

Am i missing something here?

Ysurac, 2021-09-14:
BBR2 should not be enabled on device without a 64bits CPU.

debiantippse, 2021-09-14:
> 
> 
> BBR2 should not be enabled on device without a 64bits CPU.

Okay without BBR2 it should run just fine?

Ysurac, 2021-09-14:
I have no idea, it's not a supported platform.

closed 2021-12-19 by github-actions[bot]