Ysurac/openmptcprouter #2077: OpenMPTCProuter v0.59 alpha

Ysurac, 2021-09-14:
Changes:
* VPS script Debian 11 support
* MPTCP-next kernel support on app side
* Many fixes

Only snapshot available for now: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

darthclide, 2022-02-20:
May I ask if kernel 5.15 is the one that fixes ipheth packet size problems?

Ysurac, 2022-02-20:
Patch is in both kernel.

darthclide, 2022-02-21:
Okay, will try 5.4 then since it should be more stable. Thank you for the quick response.

darthclide, 2022-02-21:
I am getting a user.notice every 4 seconds after I updated to this beta:
Mon Feb 21 01:01:29 2022 user.notice post-tracking-post-tracking: Reinstall packages...

Any ideas on what it could be?

highfreq, 2022-02-21:
VPS script is the same for both?

darthclide, 2022-02-21:
I used the command listed underneath the 5.4 release:
wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh

the version of VPS says: 0.1026 5.4.100

darthclide, 2022-02-21:
Okay I ran the command again, and it does say 1027-test now, but I am still getting that spam of "Reinstall packages...".

Ysurac, 2022-02-21:
The "Reinstall packages..." is because you did an update and a script try to reinstall packages that was installed. It was not working as I didn't make links on my server for that.
I will solve this to check less often.

darthclide, 2022-02-21:
Hmm, well the messages stopped about 5 hours ago. Were these messages a sign of something bad? Is there anything I need to do?

Ysurac, 2022-02-22:
No, I made the link on the server so packages should be installed. Nothing bad and no real problem here, only too much messages in log.

darthclide, 2022-02-22:
Ah okay thanks. No iPad wan dying on me so far. Cross your fingers that patch fixed it.

darthclide, 2022-02-23:
:/ Nope, it just went down. Are you sure the the patch listed in this link is in the 5.4 beta?:
https://github.com/openwrt/openwrt/pull/4084

If so, I have no clue how to stop big packets from shutting down the interface sigh.

darthclide, 2022-02-23:
Ignore the post I just made. I just confirmed with pings that a large MTU did not crash my iPad so that means this patch solved one issue. But this now leaves me still confused on why the iPad switches to a permanent down state until I restart it sigh.

wlatic, 2022-04-20:
Are newer versions going to be without the schedulers such as bbr2?

Ysurac, 2022-04-20:
5.4 release will keep bbr2, for 5.15 I don't know. BBR2 is made by google and for now I don't know current status (git project seems dead).

weblivepro, 2022-04-20:
I run Banana Pi R64 sdcard, after install squashfs go to run and i got:

[    3.892629] VFS: Cannot open root device "mmcblk1p65" or unknown-block(0,0): error -6
[    3.900504] Please append a correct "root=" boot option; here are the available partitions:
[    3.908867] b300        62521344 mmcblk0
[    3.908874]  driver: mmcblk
[    3.915662]   b301             512 mmcblk0p1 5452574f-2211-4433-5566-778899aabb01
[    3.915667]
[    3.924624]   b302            2048 mmcblk0p2 5452574f-2211-4433-5566-778899aabb02
[    3.924628]
[    3.933579]   b303            1024 mmcblk0p3 5452574f-2211-4433-5566-778899aabb03
[    3.933583]
[    3.942539]   b304           32768 mmcblk0p4 5452574f-2211-4433-5566-778899aabb04
[    3.942543]
[    3.951499]   b305            7168 mmcblk0p5 5452574f-2211-4433-5566-778899aabb05
[    3.951503]
[    3.960458]   b306          524288 mmcblk0p6 5452574f-2211-4433-5566-778899aabb06
[    3.960462]
[    3.969419]   103:00000        495 mmcblk0p128 5452574f-2211-4433-5566-778899aabb80

mmcblk1p65 - partition not exists - any suggestion ?

Ysurac, 2022-04-22:
·@weblivepro you use 5.4 or 5.15 image ?

ImprezyLive, 2022-04-22:
I use 5.15

I can't run correct initramfs version from SD card version, something wrong, network not starts, console not work, but i found other sd card bootloader and they running correctly initramfs.

Ysurac, 2022-04-22:
@ImprezyLive for BPI-R64 ? I looked at the code and yes it can't work for now, need some update. 5.15 is not yet tested on anything other than x86_64.

ImprezyLive, 2022-04-22:
Yes i use BPI-R64.

On this site: https://forum.banana-pi.org/t/bpi-r64-quick-start-boot-from-emmc/9809
i usese SD Card Image
And i can run initramfs without any problems, but i can't install any squashfs image.

If i use SD Card Image from you - initramfs not starting correctly, no network, console working in weird mode.

I compile from source (DEVELOP) but without successful.

finnschi, 2022-04-27:
beta 5 runs great for me, upgrade went smooth both vos and router, x86 router.

feels more solid than 0.57 at least , mire consistent speeds when uploading is where i notice it the most

evilsep, 2022-04-28:
Fresh installation on vps and rpi4, on the rpi I had to ignore the two preconfigured wans (wan1 and wan2) without deleting them and configuring new ones to use it without problems. If you need more details or logs, tell me what to do. Sorry for my English and thank you for your work.

Ysurac, 2022-04-29:
@evilsep did you create also macvlan or use some USB to ethernet key ?
Can you give me the result of ```uci show network``` or the part that begin with "network" in System->OpenMPTCProuter, "show all settings" tab ?

evilsep, 2022-04-29:
@Ysurac I use one usb to ethernet and 3 smartphones in tethering. All configured without macvlan. The image is 5.4.
The result of `uci show network`:
`network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_scheduler='blest'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fda7:9093:011b::/48'
network.globals.mptcp_version='0'
network.globals.mptcp_rr_cwnd_limited='Y'
network.globals.mptcp_rr_num_segments='1'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_syn_retries='8'
network.globals.congestion='cubic'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='6'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.device='wan1'
network.wan1.masterintf='eth1'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.multipath='off'
network.wan1.addlatency='0'
network.wan3=interface
network.wan3.device='eth1'
network.wan3.ip4table='wan'
network.wan3.defaultroute='0'
network.wan3.metric='7'
network.wan3.peerdns='0'
network.wan3.ipv6='0'
network.wan3.proto='dhcp'
network.wan3.multipath='master'
network.wan4=interface
network.wan4.device='usb0'
network.wan4.ip4table='wan'
network.wan4.multipath='on'
network.wan4.defaultroute='0'
network.wan4.metric='8'
network.wan4.peerdns='0'
network.wan4.ipv6='0'
network.wan4.proto='dhcp'
network.wan5=interface
network.wan5.device='usb1'
network.wan5.ip4table='wan'
network.wan5.multipath='on'
network.wan5.defaultroute='0'
network.wan5.metric='9'
network.wan5.peerdns='0'
network.wan5.ipv6='0'
network.wan5.proto='dhcp'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth1'
network.wan1_dev.mode='vepa'
network.wan1_dev.macaddr='a0:95:d8:6e:a1:d6'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.device='wan2'
network.wan2.masterintf='eth1'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.multipath='off'
network.wan2.addlatency='0'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth1'
network.wan2_dev.mode='vepa'
network.wan2_dev.macaddr='40:76:93:65:e8:52'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.lan_dev=device
network.lan_dev.name='eth0'
network.wan3_dev=device
network.wan3_dev.name='eth1'
network.wan4_dev=device
network.wan4_dev.name='usb0'
network.wan5_dev=device
network.wan5_dev.name='usb1'
network.wan6=interface
network.wan6.proto='dhcp'
network.wan6.device='usb2'
network.wan6.multipath='on'
network.wan6.addlatency='0'
network.wan6.defaultroute='0'
network.wan6.peerdns='0'
network.wan6.ipv6='0'
network.wan6.metric='11'`

darthclide, 2022-05-04:
Is this the VPS version I should be using for beta 5?
Version 0.1027-test 5.4.132-mptcp

darthclide, 2022-05-04:
Also in the past few betas I am seeing that my DSL gets bad ping on the status page. Before I push too much against Vultr (my VPS provider) I would like to know if there is any reason the VPS script would be causing extra random latency?

In the meantime I have tried making my own OMR-tracker that pings only the gateway, but it refuses to ping my ISP and continues to send pings to the server.

darthclide, 2022-05-07:
Just reporting that I can't put port 80 into OMR bypass without it counting down to 0 and saying "do you want to revert?"

I thought I could be cheeky and instead add http as a service but while it saved with no errors, it eventually caused luci to lock up and I could no longer access the page (ssh was fine though). Curious why it doesn't like port 80 being bypassed?

Ysurac, 2022-05-07:
@darthclide you mean in omr-bypass source port ? This conflict with OMR if you use port 80. You can still use port 443 to connect to web interface.

darthclide, 2022-05-08:
Hmmm, what if I need to redirect both port 80 and 443 to a specific interface? Am I out of luck?

Ysurac, 2022-05-08:
You can change OMR port if you want.

darthclide, 2022-05-08:
Ah well I managed to find a way around my issue. So I think it best for me not to change something you have set by default.

SLTech-SPRL, 2022-05-14:
Hi,

Packages are missing from the repositories at "https://packages.openmptcprouter.com/" for this beta release.

Personally, I just need to install open-vm-tools

For now I have manually installed the open-vm-tools package from v0.59 beta 3

Is it possible to add it ?

Thanks

TalalMash, 2022-05-21:
#2263 #2235 
Seamless failover is finally working, my settings for reference: 
VPS: Vultr France Basic ~$5
[image]
[image]
[image]

- All interfaces use MPTCP over VPN, failover is lossy without it.
- It takes 1-2 minutes to reconnect a WAN after failure or cold boot.
- ICMP/Ping shouldn't be used to test seamless failover for most cases, and ICMP is not seamless, this may cause issues for IPv6 PMTU. Selecting GTUN-UDP does not provide seamless ICMP failover for some reason.
- V2ray for TCP/UDP, GTUN-UDP for ICMP and others.
- Shadowsocks with GTUN-TCP doesn't provide seamless failover for UDP.
- GTUN-UDP is poor for UDP without proper weighing, not possible with dynamic links
- BBR2 is set for CCA

Note for first time installation: After applying all settings, restart the server first, the client second with less than a minute between. The status tab will show gateway down error in yellow, it can be safely ignored. Any changes after that can corrupt configuration on server side, you'd need to restart in the same order.
Few potential bugs:
- V2ray client-id isn't retrieved with "Force retrieve keys" 
- V2ray client-id isn't present in openmptcprouter_config.txt
- Choosing 'none' for encryption prevents v2ray from starting

hle5128, 2022-05-24:
> Patch is in both kernel.

latest beta6 on on debian 11, at the end, we got this error, just ignore?

Done
E: Version '0.1027-test' for 'omr-server' was not found
====================================================================================
OpenMPTCProuter Server 0.1027-test is now installed !
�mSSH port: 65222 (instead of port 22)
====================================================================================
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

hle5128, 2022-05-26:
the beta scrip would crash buyvm server debian 10 + 11,
after script finished, reboot, can not connect to server, interface connection lost, logged in via webconsole, no connectivity.

Ysurac, 2022-05-26:
@hle5128 don't need to had also a comment here, there is already at least 2 open issues about buyvm... And the problem is buyvm themself has they use grub without any packaged grub install... You said manual edit, so what do you change to boot ?

Ysurac, 2022-05-26:
> BPI-R64 - v0.59beta6 - running, but still the same problem with console, i type 'ifco' and enter, console no reaction, i click enter after few second, on screen i see:
> 
> **root@OpenMPTCProuter:/# ifco /bin/ash: c: not found**
> 
> v0.59beta6-5.15 and v0.59beta6-5.4 : **F0: 102B 0000 F5: 480A 0031 F5: 480A 0031 F3: 4002 0000 F2: 300C 0000 00: 1005 0000 F5: 480A 0031 F5: 480A 0031 F3: 4002 0000 F2: 300C 0000 01: 102A 0001 02: 1005 0000 BP: 0000 00C0 [0001] T0: 0000 030F [000F]** System halt!
> 
> Can't run.

I will test BPI-R64 asap.

hle5128, 2022-05-26:
> @hle5128 don't need to had also a comment here, there is already at least 2 open issues about buyvm... And the problem is buyvm themself has they use grub without any packaged grub install... You said manual edit, so what do you change to boot ?

first, it just boots directly to 5.10 after reboot,
so I have to use the web console to go to the boot menu and select kernel 5.4 mptcp
after booting to the 5.4, there is no internet connection; the connection is somehow lost regardless of which kernel after the reboot (after scripted finish); when I logged via the web console, I lost interface eth0
couldn't ping within the VM.

darthclide, 2022-05-26:
@hle5128 Just wanted to add that I noticed you tagged his response to me about a patch. Just letting you know that this is almost certainly unrelated to your problem. The patch in question has to do with ipheth errors where Apple devices somehow trigger a gateway down event even when there is full connectivity (tested by wireless ssh into the iPad and pinging any WAN IP at the same time as openwrt/openmptcprouter switches the gateway down).

@Ysurac And speaking of which, I understand you are busy so I am not going to be pushy about my niche problem, but just wanted to reiterate that this patch sadly didn't help. Maybe, just maybe, it has decreased the amount of down events, but I really am at a loss for why this error keeps showing up in the kernel:
ipheth_tx: usb_submit_urb: -16

Ysurac, 2022-05-26:
@darthclide patch is in current 5.4 (and 5.15) kernel. The error code you get is "BUSY" on USB Request Block so this doesn't seems related to previous patch.

hle5128, 2022-05-26:
> @hle5128 don't need to had also a comment here, there is already at least 2 open issues about buyvm... And the problem is buyvm themself has they use grub without any packaged grub install... You said manual edit, so what do you change to boot ?

Here is another error:

'''
Setting up libuv1:amd64 (1.34.2-1) ...
Setting up ieee-data (20180805.1) ...
Setting up python3-netaddr (0.7.19-1) ...
Processing triggers for libc-bin (2.28-10+deb10u1) ...
  Failed building wheel for uvloop
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-95__9hzx/uvloop/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-vzfo5r7g/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-95__9hzx/uvloop/
'''

at the end script.
buyvm
debian 9 and 10

darthclide, 2022-05-26:
> @darthclide patch is in current 5.4 (and 5.15) kernel. The error code you get is "BUSY" on USB Request Block so this doesn't seems related to previous patch.

Well submit_urb errors were discussed on that openwrt thread, so I figured this patch would fix it. Any suggestions?

Ysurac, 2022-05-26:
@hle5128 please don't post buyvm error here, there is already dedicated issues.
@darthclide no idea for now, can you give me a "ps aux" ? maybe multiple program try to access it at same time and make it busy, and also a "lsmod" to check if it's not a module conflict

darthclide, 2022-05-26:
The result of ps aux shows that the full version of ps isn't installed. How would I do that?
[image]

The results of lsmod:
`3c59x                  49152  0
8139cp                 32768  0
8139too                28672  0
8250_pci               49152  0
8390                   20480  1 ne2k_pci
act_csum               20480  0
act_gact               16384  0
act_ipt                16384  0
act_mirred             16384  5
act_pedit              16384  0
act_police             16384  0
act_simple             16384  0
act_skbedit            16384  0
af_key                 36864  0
ah4                    16384  0
ah6                    16384  0
ahci                   28672  0
ark3116                16384  0
arp_tables             24576  1 arptable_filter
arpt_mangle            16384  0
arptable_filter        16384  0
asix                   36864  0
asn1_decoder           16384  1 nf_nat_snmp_basic
ath                    32768  4 ath9k,ath9k_htc,ath9k_common,ath9k_hw
ath9k                 131072  0
ath9k_common           24576  2 ath9k,ath9k_htc
ath9k_htc              77824  0
ath9k_hw              360448  3 ath9k,ath9k_htc,ath9k_common
atl1                   36864  0
atl1c                  53248  0
atl1e                  40960  0
atl2                   36864  0
atm                    61440  2 solos_pci,br2684
authenc                16384  0
autofs4                45056  1
ax88179_178a           28672  0
b44                    36864  0
belkin_sa              16384  0
bnx2                   86016  0
bonding               118784  0
br2684                 20480  0
brcmfmac              270336  0
brcmutil               16384  1 brcmfmac
cdc_acm                28672  0
cdc_eem                16384  0
cdc_ether              16384  1 rndis_host
cdc_mbim               16384  0
cdc_ncm                28672  2 cdc_mbim,huawei_cdc_ncm
cdc_subset             16384  0
cdc_wdm                24576  3 cdc_mbim,qmi_wwan,huawei_cdc_ncm
cfg80211              356352 12 ath9k,ath9k_htc,ath9k_common,brcmfmac,ath,rtlwif                        i,rtl8xxxu,rtl8812au,rtl8187,rt2x00lib,mt7601u,mac80211
ch341                  16384  0
chacha_neon            20480  1 libchacha20poly1305
cls_basic              16384  0
cls_flow               20480  0
cls_fw                 16384 35
cls_matchall           16384  0
cls_route              20480  0
cls_tcindex            20480  0
cls_u32                24576 20
compat                 16384 12 ath9k,ath9k_htc,ath9k_common,rtl8192cu,rt2800usb                        ,brcmfmac,rtlwifi,rtl8xxxu,rtl8187,mt7601u,mac80211,cfg80211
compat_xtables         16384  0
cp210x                 28672  0
crc_ccitt              16384  3 rt2800lib,ppp_async,via_velocity
crc_itu_t              16384  1 mmc_spi
crc7                   16384  1 mmc_spi
[... 307 lines cut ...]
xt_hl                  16384  0
xt_iface               16384  0
xt_length              16384  0
xt_limit               16384  1
xt_mac                 16384  0
xt_mark                16384227
xt_multiport           16384114
xt_nat                 16384  0
xt_ndpi              13516801214
xt_owner               16384  0
xt_pkttype             16384  0
xt_policy              16384  0
xt_quota               16384  0
xt_recent              20480  0
xt_set                 16384145
xt_socket              16384  0
xt_state               16384  0
xt_statistic           16384  0
xt_tcpmss              16384  0
xt_tcpudp              16384 87
xt_time                16384  0
zlib_deflate           28672  1 deflate
zlib_inflate           28672  1 deflate
zram                   24576  0
zsmalloc               24576  1 zram`

Ysurac, 2022-05-27:
Yes, sorry, it's ```ps w```

hle5128, 2022-05-27:
root@OpenMPTCProuter:~# omr-iperf
iperf3: error - unable to connect to server: Resource temporarily unavailable
root@OpenMPTCProuter:~# 

omr-iperf is not working

darthclide, 2022-05-27:
results of ps w

PID USER       VSZ STAT COMMAND
    1 root      1776 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 IW<  [rcu_gp]
    4 root         0 IW<  [rcu_par_gp]
    8 root         0 IW<  [mm_percpu_wq]
    9 root         0 SW   [ksoftirqd/0]
   10 root         0 IW   [rcu_sched]
   11 root         0 SW   [migration/0]
   12 root         0 SW   [cpuhp/0]
   13 root         0 SW   [cpuhp/1]
   14 root         0 SW   [migration/1]
   15 root         0 SW   [ksoftirqd/1]
   17 root         0 IW<  [kworker/1:0H-kb]
   18 root         0 SW   [cpuhp/2]
   19 root         0 SW   [migration/2]
   20 root         0 SW   [ksoftirqd/2]
   22 root         0 IW<  [kworker/2:0H-kb]
   23 root         0 SW   [cpuhp/3]
   24 root         0 SW   [migration/3]
   25 root         0 SW   [ksoftirqd/3]
   27 root         0 IW<  [kworker/3:0H-kb]
   28 root         0 SW   [kdevtmpfs]
   29 root         0 IW<  [netns]
   31 root         0 SW   [oom_reaper]
   32 root         0 IW<  [writeback]
   33 root         0 SW   [kcompactd0]
   43 root         0 IW<  [pencrypt_serial]
   44 root         0 IW<  [pdecrypt_serial]
   57 root         0 IW<  [kblockd]
   58 root         0 IW<  [blkcg_punt_bio]
   59 root         0 SW   [watchdogd]
   62 root         0 IW<  [mptcp_wq]
   63 root         0 SW   [kswapd0]
   65 root         0 IW<  [kthrotld]
   66 root         0 SW   [irq/38-aerdrv]
   67 root         0 SW   [hwrng]
   68 root         0 IW<  [iscsi_eh]
   70 root         0 IW<  [DWC Notificatio]
   71 root         0 IW<  [uas]
   72 root         0 SW<  [vchiq-slot/0]
   73 root         0 SW<  [vchiq-recy/0]
   74 root         0 SW<  [vchiq-sync/0]
   75 root         0 IW<  [ipv6_addrconf]
   76 root         0 IW<  [kworker/u9:0]
   79 root         0 IW<  [sdhci]
   80 root         0 SW   [irq/25-mmc0]
   81 root         0 IW<  [mmc_complete]
   82 root         0 IW<  [kworker/0:1H-kb]
   83 root         0 IW<  [ext4-rsv-conver]
   88 root         0 IW<  [ata_sff]
   91 root         0 IW<  [kworker/2:1H]
  464 ubus      1260 S    /sbin/ubusd
  491 root       816 S    /sbin/askfirst /usr/libexec/login.sh
  492 root       816 S    /sbin/askfirst /usr/libexec/login.sh
  493 root         0 IW<  [kworker/0:2H]
  495 root         0 IW<  [kworker/1:1H-kb]
  535 root      4032 S    /sbin/rngd -f -r /dev/urandom
  559 root         0 IW<  [kworker/3:2H-kb]
  826 root         0 IW<  [bond0]
 1191 root         0 IW<  [cfg80211]
 1208 root      2404 S    {ntpd} /sbin/ujail -n ntpd -U ntp -G ntp -C /etc/capab                                      ilities/ntpd.json -c -u -r /bin/ubus -r /usr/bin/en
 1218 ntp       1288 S    /usr/sbin/ntpd -n -N -l -S /usr/sbin/ntpd-hotplug -p 0                                      .pool.ntp.org -p 1.pool.ntp.org -p 2.pool.ntp.org -
 1879 logd      1264 S    /sbin/logd -S 64
 1948 root      2344 S    /sbin/rpcd -s /var/run/ubus/ubus.sock -t 120
 1965 root      1328 S    /usr/sbin/glorytun keyfile /tmp/glorytun-vpn.key port                                       65001 host 149.28.122.251 dev tun0 mptcp chacha20 r
 2044 root         0 IW   [kworker/0:2-eve]
 2252 root      1012 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22                                       -K 300 -T 3
[... 36 lines cut ...]
 9469 root         0 IW   [kworker/0:1-mm_]
 9882 root         0 IW   [kworker/u8:1-ev]
10691 unbound  38412 S    /usr/sbin/unbound -d -c /var/lib/unbound/unbound.conf
12469 root         0 IW   [kworker/1:1-eve]
12632 root      2404 S    {dnsmasq} /sbin/ujail -n dnsmasq -u -l -r /dev/null -r                                       /dev/urandom -r /etc/TZ -r /etc/dnsmasq.conf -r /e
12634 dnsmasq   2928 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -                                      k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
13101 root         0 IW   [kworker/3:1-mm_]
13638 root         0 IW   [kworker/0:0-mm_]
13745 root         0 IW   [kworker/2:0-mm_]
14060 root         0 IW   [kworker/3:2-eve]
14485 root      1288 S    udhcpc -p /var/run/udhcpc-eth1.pid -s /lib/netifd/dhcp                                      .script -f -t 0 -i eth1 -x hostname:OpenMPTCProuter
14859 root         0 IW   [kworker/1:0-eve]
16864 root         0 IW   [kworker/u8:0-mp]
17005 root      1288 S    udhcpc -p /var/run/udhcpc-usb0.pid -s /lib/netifd/dhcp                                      .script -f -t 0 -i usb0 -x hostname:OpenMPTCProuter
17062 root         0 IW   [kworker/u8:4-mp]
19366 root      1288 S    udhcpc -p /var/run/udhcpc-usb1.pid -s /lib/netifd/dhcp                                      .script -f -t 0 -i usb1 -x hostname:OpenMPTCProuter
19598 root         0 IW   [kworker/u8:3-mp]
20150 root         0 IW   [kworker/2:1-eve]
21822 root      2472 S    /usr/bin/ss-local -c /var/etc/shadowsocks-libev/ss_loc                                      al.tracker.json
21823 root      2472 S    /usr/bin/ss-local -c /var/etc/shadowsocks-libev/ss_loc                                      al.tracker_sss0.json
21825 root     12004 S    /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_red                                      ir.hi.json
21826 root      2280 S    /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_red                                      ir.hi2.json
21828 root     12800 S    /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_red                                      ir.hi4.json
22067 root      5204 S    /usr/sbin/usbmuxd --systemd
26692 root         0 IW   [kworker/u8:2-mp]

darthclide, 2022-05-29:
Woke up today to see that it wasn't just gateway down, but the whole interface was gone with 0 information. Restarting it through luci fixed it, but I looked through the logs to see when it went down today. This is what I found:

> Sun May 29 16:35:09 2022 kern.info kernel: [1928212.744923] usb 1-1.3.4: USB disconnect, device number 14
Sun May 29 16:35:09 2022 daemon.notice netifd: Interface 'wan5' is disabled
Sun May 29 16:35:09 2022 kern.info kernel: [1928212.791616] ipheth 1-1.3.4:4.2: Apple iPhone USB Ethernet now disconnected
Sun May 29 16:35:09 2022 daemon.err usbmuxd[22067]: [16:35:09.419][3] Removed device 2 on location 0x1000e
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.003711] usb 1-1.3.4: new high-speed USB device number 15 using xhci_hcd
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.113310] usb 1-1.3.4: New USB device found, idVendor=05ac, idProduct=12ab, bcdDevice= 7.12
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.122021] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.129693] usb 1-1.3.4: Product: iPad
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.133626] usb 1-1.3.4: Manufacturer: Apple Inc.
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.138504] usb 1-1.3.4: SerialNumber: be8acf2681e229a84965305c8e1159bdddf56545
Sun May 29 16:35:21 2022 kern.debug kernel: [1928225.146339] usb 1-1.3.4: no of_node; not parsing pinctrl DT
Sun May 29 16:35:21 2022 kern.debug kernel: [1928225.169296] ipheth 1-1.3.4:4.2: no of_node; not parsing pinctrl DT
Sun May 29 16:35:21 2022 daemon.notice netifd: Interface 'wan5' is enabled
Sun May 29 16:35:21 2022 kern.info kernel: [1928225.176195] ipheth 1-1.3.4:4.2: Apple iPhone USB Ethernet device attached
Sun May 29 16:35:21 2022 daemon.err usbmuxd[22067]: [16:35:21.805][3] Connecting to new device on location 0x1000f as ID 3
Sun May 29 16:35:21 2022 daemon.err usbmuxd[22067]: [16:35:21.805][3] Connected to v2.0 device 3 on location 0x1000f with serial number be8acf2681e229a84965305c8e1159bdddf56545
Sun May 29 16:35:22 2022 user.notice OMR-Rename: modalias: usb:v13B1p0041d3000dc00dsc00dp00icFFiscFFip00in00-24:f5:a2:f2:0e:9f - chk_modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - ifname: eth1 - INTERFACE: eth3
Sun May 29 16:35:22 2022 user.notice OMR-Rename: modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - chk_modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - ifname: eth2 - INTERFACE: eth3
Sun May 29 16:35:22 2022 user.notice OMR-Rename: Rename eth3 to eth2
Sun May 29 16:35:22 2022 user.notice OMR-Rename: modalias: usb:v22B8p2E25d0409dc00dsc00dp00icE0isc01ip03in00-52:fc:2c:dc:92:b9 - chk_modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - ifname: usb0 - INTERFACE: eth3
Sun May 29 16:35:22 2022 user.notice OMR-Rename: modalias: usb:v22B8p2E25d0409dc00dsc00dp00icE0isc01ip03in00-7a:85:3a:54:25:4d - chk_modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - ifname: usb1 - INTERFACE: eth3
Sun May 29 16:35:22 2022 user.notice OMR-Rename: modalias: usb:v1410pB020d0318dc00dsc00dp00icEFisc04ip01in00-00:15:ff:39:96:37 - chk_modalias: usb:v05ACp12ABd0712dc00dsc00dp00icFFiscFDip01in02-fe:1d:43:49:46:32 - ifname: eth3 - INTERFACE: eth3
Sun May 29 16:35:25 2022 daemon.info [6494]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.4': not supported by any plugin

What I find interesting is that part where it tried to rename it from eth3 to eth2 even though eth2 is my Verizon connection and should be off limits? My verizon always sits around 2 bars of strength, so maybe it is bad luck that my AT&T went down at the same time as my Verizon? And it thought eth2 was available?

winkmichael, 2022-06-18:
Really enjoying 0.59 beta 7. Running on Pi4 with 5.4 kernel and VPS from OVH on Debian 10. Speeds are improved over the current release and I no longer get any glitching where parts of webpages fail to come in. Great work & thank you!

ImprezyLive, 2022-06-29:
Debian 10
Update failed, omr-glorytun (0.3.4-4) 

Setting up omr-glorytun (0.3.4-4) ...
Job for glorytun-udp@tun0.service failed because the control process exited with error code.
See "systemctl status glorytun-udp@tun0.service" and "journalctl -xe" for details.
dpkg: error processing package omr-glorytun (--configure):
 installed omr-glorytun package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-glorytun
E: Sub-process /usr/bin/dpkg returned an error code (1)

Any solution?

Ysurac, 2022-06-29:
@ImprezyLive It's an update or a fresh install ? For 5.4 or 5.15 ? I would need the result of "journalctl -xe" on the VPS

ImprezyLive, 2022-06-29:
System Update (old version 1022).

Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Start-post operation timed out. Stopping.
Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Control process exited, code=killed, status=15/TERM
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStartPost= process belonging to unit glorytun-udp@tun0.service has exited.
--
-- The process' exit code is 'killed' and its exit status is 15.
Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit glorytun-udp@tun0.service has entered the 'failed' state with result 'exit-code'.
Jun 29 13:23:16 Mikolaj1 systemd[1]: Failed to start Glorytun UDP on tun0.
-- Subject: A start job for unit glorytun-udp@tun0.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit glorytun-udp@tun0.service has finished with a failure.
--
-- The job identifier is 5978 and the job result is failed.
Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Scheduled restart job, restart counter is at 1.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit glorytun-udp@tun0.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jun 29 13:23:16 Mikolaj1 systemd[1]: Stopped Glorytun UDP on tun0.
-- Subject: A stop job for unit glorytun-udp@tun0.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit glorytun-udp@tun0.service has finished.
--
-- The job identifier is 6128 and the job result is done.
Jun 29 13:23:16 Mikolaj1 systemd[1]: Starting Glorytun UDP on tun0...
-- Subject: A start job for unit glorytun-udp@tun0.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit glorytun-udp@tun0.service has begun execution.
--
-- The job identifier is 6128.
Jun 29 13:23:16 Mikolaj1 glorytun-udp-run[25807]: error: `to' is not a valid IPADDR for bind
Jun 29 13:23:16 Mikolaj1 systemd[1]: glorytun-udp@tun0.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit glorytun-udp@tun0.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.

ImprezyLive, 2022-06-29:
Is something wrong with update process.
After install on new server, everything is ok.

Ysurac, 2022-06-29:
The issue should be fixed, it was due to update from an old release.

ngarafol, 2022-07-07:
@Ysurac typo in vps script for debian 11:

There is no `ifconfig` in debian 11, so line 120 of `/usr/local/bin/multipath` should be changed to:

```
# grep -n "ip a sh" /usr/local/bin/multipath 
120:        IP=$(ip a sh $DEVICE | sed -En 's/127.0.0.1//;s/.*inet (addr:)?(([0-9]*\.){3}[0-9]*).*/\2/p')
```
Also, for some strange reason using beta7 0.59 on x86_64, ping and dns does not work for me (from local omr, vps is working fine).

Ysurac, 2022-07-07:
@ngarafol I fixed the script. It's the same script that is used for all debian and ubuntu releases.

Can you test beta 8 for your DNS/Ping problem ? (I think it's a VPN issue)

ngarafol, 2022-07-07:
In the meantime, just to confirm beta7 5.15 omr + debian 10 vps with 5.15 works (ping and dns both local and vps side). 
local: `Version 0.59beta7-5.15 `
vps: debian 10 + `Version 0.1027-test 5.15.41-mptcp`

dooley1981, 2022-07-10:
what is is benefit of using one kernel 5.15 over the other 5.4?

Ysurac, 2022-07-10:
No real benefit for now. Kernel 5.15 seems to give slower speed in my tests.
MPTCP in kernel 5.15 is the upstreamed release and it's actively maintained by more people.

dooley1981, 2022-07-10:
Okay thanks for the reply.

[39 more replies]

closed 2023-01-04 by github-actions[bot]