Ysurac/openmptcprouter #2078: Significant performance impact to having WAN devices on tagged VLAN

gpontis, 2021-09-14:
## Expected Behavior
Connecting WAN devices through a single ethernet port with VLAN tagging should perform similarly to direct connect using multiple ethernet ports

## Current Behavior
Performance is significantly reduced in VLAN configuration

## Possible Solution
unknown

## Steps to Reproduce the Problem

  1. Connect two or more modems to a switch
  2. Configure the switch to tag the packets from each modem. One VLAN per modem
  3. Configure an addition port on the switch to provide access to all the modem VLANs ( tagged ) and connect it to a port on the OMR device.
  4. Configure OMR WAN interfaces to match modem IP and VLAN, all referencing the same physical port
  5. Connect computer to LAN port of OMR and run speed test
  6. Reconfigure OMR with modems directly connected and repeat speed test 

## Context (Environment)
In my case the modems are somewhat remotely located, and a second computer may be connected to them for monitoring and configuration. Being able to use a single cable to interface to the modems from anywhere in the network is useful. It also makes it easy to power the modems if the switch has PoE capability. Lastly, it reduces the number of physical ports required on the OMR device.

For some actual numbers, I tested with two DSL modems and a 4G modem. The DSL modems sync at 15/1.25 and the 4G delivers something around 5000 down / 5000 up. Using the VLAN, the aggregate throughput is around 11000/1100. ( All speeds in Kb/s ). Without using the VLAN, the aggregate throughput is 27000/6000.

I verified that it is not a switch problem by adding an additional switch in front of the OMR box that took the same tagged feed of three modems on three VLANs, and separated them into the different untagged ports. Reconfiguring OMR to use three different WAN interfaces, and plugging them into the three switch ports restored full performance.

Command-line "top" does not show high utilization. In fact at a glance all tasks are at 0 or 1%.

I also briefly experimented with MTU sizes on the interfaces but could not affect the results.

## Specifications

  - OpenMPTCProuter version: 0.58.4
  - OpenMPTCProuter VPS version: 0.58.4
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: x86-64, Protectli FW4B with 4 core celeron
  - Country: USA

Adorfer, 2021-09-15:
this confirms my observation: 
If you want best performance ("linespeed"), use dedicated NICs (and not USB-eths) for each WAN connection. 

(this may be the same thing that makes it difficult for Raspi4 to saturate 1000TX above 500MBit/s despite the physical hardware beeing a full duplex port on the PCIe)

gpontis, 2021-09-15:
An often recommended connection is a single NIC, as I tested, but using MacVLAN instead of traditional IEEE 802.1Q VLANs. If that can give good performance then I think that VLANs should also be good.

Geo.

closed 2021-12-20 by github-actions[bot]