Ysurac/openmptcprouter #2079: Turn off Rate Limiting on piHole

finnschi, 2021-09-16:
**Is your enhancement proposal related to a problem? Please describe.**

I am being instantely rate limited because to piHole on the VPS my whole network is 1 single client. 
**Describe the solution you'd like**

 setting "RATE_LIMIT=0/0" in /etc/pihole/pihole-FTL. conf 
**Describe alternatives you've considered**

No idea, its a very simple fix :D 
**Additional context**

hairy-dog, 2021-09-25:
This was also an issue with a pihole server on the lan. The OMR Router was being rate limited.

Starsick, 2022-01-14:
I am having a similar error message. How do you (and should I) turn off rate limiting? Here is the message I get in the GUI:
[image]
I would need step by step instructions as I am not a programmer by any means.

ihavenokia, 2022-01-18:
> I am having a similar error message. How do you (and should I) turn off rate limiting? Here is the message I get in the GUI: [image] I would need step by step instructions as I am not a programmer by any means.

I'm having the same problem

ihavenokia, 2022-01-18:
noice! thank you guys <3

pylaterreur, 2026-10-04:
With Pi-hole v6 this is back. `omr-pihole.sh` (VPS develop 0a59fca) still writes `RATE_LIMIT=0/0` to `/etc/pihole/setupVars.conf`, but the Pi-hole installer now sets up v6 (Core v6.4.3, FTL v6.7.1 today). v6 keeps all its settings in `/etc/pihole/pihole.toml` and doesn't read `setupVars.conf` or `/etc/dnsmasq.d` any more. It has no lighttpd either, so #2257 doesn't apply to it.

I tested it in a Debian 13 VM: a fresh Pi-hole v6, then the OMR part of `omr-pihole.sh`, with a "router" at 10.255.240.2 on a tunnel-like interface (VPS side 10.255.240.1):
- The script prints `cannot create /etc/lighttpd/external.conf: Directory nonexistent` and two grep errors. `pihole-FTL --config dns.rateLimit.count` is still 1000 (per 60 s).
- Of 1100 queries sent by the router in 4.6 s, 1000 were answered and 100 got REFUSED, and FTL.log has `Rate-limiting 10.255.240.2 for at least 49 seconds`.

The `setupVars.conf` it creates also breaks DNS at the next Pi-hole update. When `pihole -up` updates the core or FTL, it runs the installer, whose `migrate_dnsmasq_configs` takes any install with a `setupVars.conf` for v5 and runs `pihole-FTL migrate v6` (so does running `omr-pihole.sh` again). That rebuilds `pihole.toml` from the file: defaults plus `dns.interface = gt-tun0`, with `dns.upstreams = []`. After that, every query is REFUSED.

The patch below sets the rate limit with `pihole-FTL --config` and drops the v5-only part. Before the installer runs, it removes a `setupVars.conf` left by an earlier run (only when `pihole.toml` exists). Pi-hole v6's default listening mode (`LOCAL`) answers queries from the subnets of all the tunnels, MQVPN and WireGuard included, and the VPS firewall keeps port 53 closed on the WAN. With the patch, the rate limit is 0/0, all 1100 queries are answered, and the upstream servers survive a simulated update.

On a VPS where the script already ran with Pi-hole v6, this undoes it:
```
rm /etc/pihole/setupVars.conf
pihole-FTL --config dns.rateLimit.count 0
pihole-FTL --config dns.rateLimit.interval 0
```

<details><summary>Patch against openmptcprouter-vps develop 0a59fca (<code>git am</code>)</summary>

```diff
From 6e30db0dda704685647732b51d2d8ba80e8ce2ef Mon Sep 17 00:00:00 2001
From: Pierre-Yves Le Borgne <pylaterreur@gmail.com>
Date: Sun, 4 Oct 2026 09:06:18 +0100
Subject: [PATCH] omr-pihole: configure Pi-hole v6

The Pi-hole installer now sets up v6, which keeps its settings in
/etc/pihole/pihole.toml and no longer reads setupVars.conf or
/etc/dnsmasq.d, and has no lighttpd. The OMR part of the script failed on
the missing files and changed nothing:

- RATE_LIMIT=0/0 was ignored, so the router, which sends the queries of
  the whole LAN from its tunnel address, is limited to 1000 queries a
  minute, then gets REFUSED for the rest of the minute.
- The setupVars.conf it created made the next Pi-hole update take the
  install for v5 and migrate it: pihole.toml is rebuilt from that file,
  without the upstream DNS servers, and every query is REFUSED.

Turn the rate limit off with pihole-FTL --config, leave setupVars.conf
alone, and remove one left by an earlier run before the installer runs.
Pi-hole v6's default listening mode answers the tunnels' subnets, and the
VPS firewall keeps port 53 closed on the WAN.
---
 omr-pihole.sh | 40 ++++++++++++++--------------------------
 1 file changed, 14 insertions(+), 26 deletions(-)

diff --git a/omr-pihole.sh b/omr-pihole.sh
index 928f3d9..f5fdb7f 100644
--- a/omr-pihole.sh
+++ b/omr-pihole.sh
@@ -19,37 +19,25 @@ echo "Don't apply Pi-hole firewall rules."
 echo "!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!"
 [ "`tty`" != "not a tty" ] && read -n 1 -s -r -p "Press any key to continue" || sleep 5
 
+# A setupVars.conf next to pihole.toml can only come from an earlier run of
+# this script on Pi-hole v6. The installer would take it for a v5 install to
+# migrate, and rebuild pihole.toml from it: the upstream DNS servers are lost.
+[ -f /etc/pihole/pihole.toml ] && rm -f /etc/pihole/setupVars.conf
+
 echo "Run Pi-hole install script..."
 curl -sSL https://install.pi-hole.net | bash
 echo "Done"
 echo "-------------------------------------------------------------------------------------------------------------------------------"
 echo "OMR Pi-hole configuration..."
-cat > /etc/lighttpd/external.conf << 'EOF'
-server.bind="10.255.255.1"
-$SERVER["socket"] == "10.255.254.1:80" { }
-$SERVER["socket"] == "10.255.252.1:80" { }
-$SERVER["socket"] == "10.255.251.1:80" { }
-$SERVER["socket"] == "10.255.253.1:80" { }
-EOF
[... 12 lines cut ...]
-cat > /etc/dnsmasq.d/99-omr.conf <<-EOF
-interface=gt-tun0
-interface=gt-udp-tun0
-interface=tun0
-interface=mlvpn0
-interface=dsvpn0
-EOF
+# Pi-hole v6, what its installer sets up now, keeps all its settings in
+# /etc/pihole/pihole.toml: it no longer reads setupVars.conf or
+# /etc/dnsmasq.d, and has no lighttpd. Its default listening mode answers
+# the subnets of all the tunnels, and the VPS firewall keeps port 53 closed
+# on the WAN.
+# The router sends the queries of the whole LAN from its tunnel address, so
+# turn off the rate limit (by default 1000 queries a minute per client).
+pihole-FTL --config dns.rateLimit.count 0 >/dev/null
+pihole-FTL --config dns.rateLimit.interval 0 >/dev/null
 systemctl -q restart pihole-FTL
 echo "Done"
 echo "======================================================================================================================================"
-- 
2.51.0

```

</details>

pylaterreur, 2026-10-04:
One correction to the patch above, before it's applied.

My `omr-pihole.sh` patch removes a leftover `setupVars.conf` (when `pihole.toml` is present) so a later `pihole -up` doesn't take the install for v5 and migrate it. That part is right, but on its own it would stop the router from using the VPS Pi-hole, because of a second, pre-existing bug on the VPS side:

`omr-admin` decides whether Pi-hole is installed by the presence of **`/etc/pihole/setupVars.conf`** only (`omradmin.py`, in the `/config` handler):

```python
if os.path.isfile('/etc/pihole/setupVars.conf'):
    pihole = True
else:
    pihole = False
```

Pi-hole v6 has no `setupVars.conf` — its migration moves that file to `/etc/pihole/migration_backup_v6/` and keeps everything in `/etc/pihole/pihole.toml`. So today a VPS whose Pi-hole was updated to v6 already reports `pihole.state = false`, and the router's `_get_pihole`/`_get_vps_config` (feeds `openmptcprouter-vps`) never points dnsmasq at the VPS. My `omr-pihole.sh` change, by removing the stale `setupVars.conf`, would make that true for a fresh v6 install too.

The fix is to have `omr-admin` also accept `pihole.toml`, which is what Pi-hole's own installer does to detect an existing install (`check_fresh_install` tests `pihole.toml` **or** `setupVars.conf`). I picked this over having `omr-pihole.sh` drop a marker file: it needs no extra state, it tracks Pi-hole's own notion of "installed", and it fixes the already-broken v6 case rather than only new installs.

Tested with the vps-admin unit suite (pytest, Python 3.11 as the repo's CI runs it): a `/config` request reports `pihole.state` true when only `pihole.toml` is present, true when only `setupVars.conf` is present, true when both are, and false when neither is (or only the migrated `migration_backup_v6/setupVars.conf` exists). Full suite green.

<details><summary>Patch against openmptcprouter-vps-admin develop 84b20c1 (<code>git am</code>)</summary>

```
From 33c2d2032f5c8a80bb7f740873397bea86f39775 Mon Sep 17 00:00:00 2001
From: Pierre-Yves Le Borgne <pylaterreur@gmail.com>
Date: Sun, 4 Oct 2026 19:38:50 +0100
Subject: [PATCH] omr-admin: report Pi-hole v6 as installed in /config

/config set pihole.state only when /etc/pihole/setupVars.conf exists.
Pi-hole v6 keeps its settings in /etc/pihole/pihole.toml and has no
setupVars.conf: its migration from v5 moves the file to
/etc/pihole/migration_backup_v6/. So a VPS whose Pi-hole was updated to
v6 reports pihole.state false, and so would a v6 install set up by
omr-pihole.sh once that script stops writing a setupVars.conf (which
makes Pi-hole updates migrate the install again and drop its upstream
servers). With pihole.state false, the router stops sending its DNS
queries to the VPS Pi-hole.

Take either file as "Pi-hole is installed", as Pi-hole's own installer
does (check_fresh_install). "pihole uninstall" removes /etc/pihole, so
an uninstalled Pi-hole is still reported as not installed.
---
 omradmin.py      |  5 ++++-
 test/test_api.py | 24 ++++++++++++++++++++++++
 2 files changed, 28 insertions(+), 1 deletion(-)

diff --git a/omradmin.py b/omradmin.py
index 0283475..d5cac84 100755
--- a/omradmin.py
+++ b/omradmin.py
@@ -2960,7 +2960,10 @@ async def config(userid: Optional[int] = Query(None), username: Optional[str] =
     else:
         iperf3_key = ''
 
-    if os.path.isfile('/etc/pihole/setupVars.conf'):
+    # Pi-hole v6 keeps its settings in pihole.toml, and has no setupVars.conf
+    # (its migration from v5 moves that file away). Pi-hole's own installer
+    # takes either file as "Pi-hole is installed".
+    if os.path.isfile('/etc/pihole/pihole.toml') or os.path.isfile('/etc/pihole/setupVars.conf'):
         pihole = True
     else:
         pihole = False
diff --git a/test/test_api.py b/test/test_api.py
index 407581a..f4ea710 100644
--- a/test/test_api.py
+++ b/test/test_api.py
@@ -325,6 +325,30 @@ class TestConfig:
         assert r.status_code == 200
 
 
+class TestConfigPihole:
+    """/config must report Pi-hole v6 as installed.
+
+    The router uses the VPS Pi-hole only when pihole.state is true. Pi-hole
+    v6 keeps its settings in pihole.toml and has no setupVars.conf: its
+    migration from v5 moves that file to migration_backup_v6/.
+    """
+
+    @pytest.mark.parametrize("files, state", [
+        pytest.param(("/etc/pihole/pihole.toml",), True, id="v6"),
+        pytest.param(("/etc/pihole/setupVars.conf",), True, id="v5"),
+        pytest.param(("/etc/pihole/pihole.toml", "/etc/pihole/setupVars.conf"),
+                     True, id="both"),
+        pytest.param(("/etc/pihole/migration_backup_v6/setupVars.conf",),
+                     False, id="v5-backup-only"),
+        pytest.param((), False, id="none"),
+    ])
+    def test_pihole_state(self, user_client, files, state):
+        with patch("os.path.isfile", side_effect=_isfile_for(*files)):
+            r = user_client.get("/config")
+        assert r.status_code == 200
+        assert r.json()["pihole"]["state"] is state
+
+
 # ===========================================================================
 # Shadowsocks
 # ===========================================================================
-- 
2.51.0

```

</details>

So the two go together: the `omr-pihole.sh` patch above, plus this one-liner in `omr-admin`.

closed 2026-10-05 by Ysurac