Ysurac/openmptcprouter #2081: the load does not get seperated into multiple lines

rifaterdemsahin, 2021-09-17:
## Expected Behavior
I expect the load on each line. 
I should have 700 Down and 200 up

## Current Behavior
[image]
[image]

## Possible Solution
Separate the load on each so one line does not go down

## Steps to Reproduce the Problem

  1. add 150 GB file in the network to sync it to the cloud
  2. Everything gets slowed down
  

## Context (Environment)
The internet has become not much slower.

## Specifications
don't know why it says proxy disabled.

[image]

Ysurac, 2021-09-17:
Please use the template, I need versions, platform,...
Proxy is disabled mean the proxy was disabled in advanced settings, in the wizard.
Without proxy this can't work correctly.

rifaterdemsahin, 2021-09-17:
vps-ea15b92c (51.195.1xxxxxx)
Version 0.1023 5.4.100-mptcp
(Latest available version 0.1026)
Load: 0.76 0.67 0.57
 

OpenMPTCProuter 
(192.168.100.1)Proxy is DISABLEDVersion 0.58.5Load: 0.30 0.29 0.31Uptime: 41d 18h 23m 22slan address: 192.168.100.1lan range: 192.168.100.100 - 192.168.100.249

Is there a practical script/page where I can get all these information ? 
When I copy-paste ...it brings the html.

rifaterdemsahin, 2021-09-17:
I had the old proxy server settings there for backup now removed it
[image]

now after removing the backup config I am back into action
I suspect that disabled server had a faulty code... which started affecting the system after 40 days...or after I updated the text.

now I remembered I updated the text...and I had a disabled server there...so there needs to be a unittest for that case.

[image]

[image]

rifaterdemsahin, 2021-09-17:
[image]

now i see all the lines active
i need to have an open window for this

closed 2021-12-21 by github-actions[bot]