Ysurac/openmptcprouter #2082: Possible issue with omr-test-speed on targeting specific interfaces

donbecker, 2021-09-18:
## Expected Behavior
I'd like to test specific interfaces with omr-test-speed, however it appears I can only do so to the interface of my multipath master

## Current Behavior
Working: 
Test all connections at once: `omr-test-speed`
Test my multipath master interface only: `omr-test-speed eth5`

Not working (fails to ping any during "Select best test server..." step): 
Test my multipath enabled interface only: `omr-test-speed eth3`
Test my multipath enabled interface only: `omr-test-speed eth2`

## Possible Solution
I'm not sure where in the code this error is. 

## Steps to Reproduce the Problem

(please see above commands)

## Context (Environment)
I'm trying to individually test connection speed of each interface. 

## Specifications

  - OpenMPTCProuter version: v0.58
  - OpenMPTCProuter VPS version: v0.1026.5.4.100-mptcp
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: x86_64
  -   - Country: US

Ysurac, 2021-09-18:
omr-test-speed work with any interface.
Ping is blocked on some connection ?
What is the result of "ip r" ? All is green in status page ?

donbecker, 2021-09-20:
On the status page, only my master connection is green status. 

The other two ("multipath: on / enabled") are red "No Server ping response after 1 second". 

If I SSH to the router, I can ping all three fine. 

Let me know if there are any commands to run to further troubleshoot. I'm not sure how  you want me to run `ip -r`

closed 2021-12-24 by github-actions[bot]